betaflight/betaflight-configurator #3524: GPS waiting info in wrong position

TheIsotopes, 2023-07-22:
### Describe the bug

see the screenshot

[image]

### To Reproduce

using latest configurator 10.10

### Expected behavior

info in right position

### Configurator version

10.10

haslinghuis, 2023-07-22:
Shows like this on Linux:

[image]

[image]

[image]

Margin-top needs to be 30% for 1920*1080 instead of 50% (tested on 1280*1024)

[image]

TheIsotopes, 2023-07-22:
ok. with `margin-top: 30%;` it looks good 👍

HThuren, 2023-07-22:
Good idea, I have never seen this text :-)

PR #3525 (2023-07-23): Fix GPS waiting info margin
Fixes: #3524

Added some colors in second commit:

[image]
closed 2023-07-23 by haslinghuis