betaflight/betaflight-configurator #3528: make adjusting frame lines easier

malinhiles, 2023-07-25:
### Is your feature request related to a problem? Please describe

[image]

When configuring this "camera frame" osd element, it is easy to move an locate the camera frame, but difficult to change the size. Changing the size of the camera frame is pretty clunky right now. It requires guessing and checking through the cli, which requires a reboot every time.

AN improvement here will appreciated most by cinema fpv pilots who frequently fly cameras with multiple lens configurations. Changing to a different lens currently requires fumbling around in the cli, guessing and checking, multiple reboots from the cli.

### Describe the solution you'd like

The full solution here has two parts.
1. When the camera frame element is enabled, have corresponding settings for size appear in the right panel of the osd tab.
Over here:
[image]

2. Make the size and position of these on screen elements accessible through the OSD part of the stick menu. This would mean that something as simple as swapping a lens on your payload camera doesn't also require a laptop plugged in and minutes spent guess and checking in the cli.

### Describe alternatives you've considered

NA

malinhiles, 2023-08-26:
Any thoughts on this?

malinhiles, 2023-09-26:
any ideas?

yelmarb, 2023-11-17:
This feature would be especially helpful for shooting 9:16 format videos for mobile social media platforms.

malinhiles, 2023-11-17:
> This feature would be especially helpful for shooting 9:16 format videos for mobile social media platforms.

@yelmarb that too!

yelmarb, 2023-11-17:
Would also be great to adjust the opacity of the framing lines as well.