betaflight/betaflight-configurator #3546: UI range for `gps_rescue_descent_dist` is incorrect in `master`

nerdCopter, 2023-08-14:
### Describe the bug

please see closed PR https://github.com/betaflight/betaflight-configurator/pull/3542 .

Value range for `gps_rescue_descent_dist` is incorrect in Configurator UI. 

I will work on this single item... seems MSP version 1.45 is where `10` began to be used rather than `30`. (ref: https://github.com/betaflight/betaflight/pull/11579).

also found `5` was used for a stint: https://github.com/betaflight/betaflight/pull/11834 -- MSP was already version 1.45 here. i prefer not to add too much complexity and will not accommodate `5`.  firmware reverted to `10` in https://github.com/betaflight/betaflight/pull/12715 .

We may need to ask @ctzsnooze and @freasy for a complete list of needed UI ranges and what MSP they start.

### To Reproduce

n/a

### Expected behavior

n/a

### Configurator version

`master`

haslinghuis, 2023-08-14:
See https://github.com/betaflight/betaflight-configurator/pull/3544

If there a more values changed please let me know

nerdCopter, 2023-08-14:
> See #3544
> 
> If there a more values changed please let me know

i did not investigate, but presume there are.  i feel like GPS devs should compile what is needed to save from blind-searching.

PR #3544 (2023-08-14): Fix gps_rescue_min_dth
Closes #3542
Closes #3546
closed 2023-08-14 by haslinghuis