betaflight/betaflight-configurator #3553: Chrome API Error: Failed to connect to the port..

Bwsbmw, 2023-08-22:
### Describe the bug

Im having a hell of a time with betaflight 10.9.0 on a new Windows 11 Pro Desktop. Every betaflight FC i plug in, it will not Auto-Connect. I can click Connect and it will connect to the boards just fine. None of the board will go from the CLI. and Save then enter. It doesnt reconnect, i get Chrome API Error: Failed to connect to the port.. and have to manually click the Connect button. Same thing with DFU mode. On a windows 10 pc, the same FC's, with the same USB cable will Auto Connect, and Enter DFU mode. 

### To Reproduce

Connect FC to Computer Via USB. Auto connect wont load, Chrome API Error: Failed to connect to the port.. is in the log. only way to connect to FC is manually clicking CONNECT.

### Expected behavior

Auto Connect upon pluggin in FC. Go into DFU mode even when Holding the button. I know i have to be missing something, oddly everything else seems to work just fine threw USB. i have also tried the USB ports on the actual motherboard with not sucsess. although i can manually connect after i get the Chrome API Error: Failed to connect to the port..

### Configurator version

10.9.0 697943b

### Add any other context about the problem that you think might be relevant here

Same FC's and USB cable work fine on Windows 10 pc.

haslinghuis, 2023-08-22:
Please try [nightly build](https://github.com/betaflight/betaflight-configurator-nightlies/releases)

Bwsbmw, 2023-08-22:
No change, tried 2 fc's, still have to Click Connect manually. Wont reconnect from CLI either after SAVE then Enter.
2023-08-21 @21:24:24 -- OS: Windows
2023-08-21 @21:24:24 -- Configurator: 10.10.0-debug-52df6a1
2023-08-21 @21:24:59 -- Chrome API Error: Failed to connect to the port..
2023-08-21 @21:24:59 -- Failed to open serial port
Clicked Connect
2023-08-21 @21:25:10 -- Serial port successfully opened with ID: 1
2023-08-21 @21:25:10 -- MultiWii API version: 1.44.0
2023-08-21 @21:25:10 -- Flight controller info, identifier: BTFL, version: 4.3.0
2023-08-21 @21:25:10 -- Running firmware released on: Mar 29 2021 10:51:53
2023-08-21 @21:25:10 -- Board: AIRB/OMNIBUSF4SD(STM32F405), version: 0
2023-08-21 @21:25:10 -- Unique device ID: 0x24001d3536471334353030
2023-08-21 @21:25:10 -- Craft name:
2023-08-21 @21:25:10 -- Arming Disabled
2023-08-21 @21:25:14 -- CLI mode detected

haslinghuis, 2023-08-22:
F12 output?

Bwsbmw, 2023-08-22:
[ajdoecjhplhoahggfmgflfnlechcljoc-1692668274355.log](https://github.com/betaflight/betaflight-configurator/files/12403837/ajdoecjhplhoahggfmgflfnlechcljoc-1692668274355.log)

Bwsbmw, 2023-08-22:
Replacing timeout/interval functions with Node versions
localization.js:209 Detected locale en-US
i18next.js:22 i18next::backendConnector: loaded namespace messages for language en Object
i18next.js:22 i18next: languageChanged en
i18next.js:22 i18next: initialized Object
init.js:32 i18n initialized, starting Vue framework
localization.js:46 i18n system loaded
localization.js:209 Detected locale en-US
main.js:225 Libraries: jQuery - 3.7.0, d3 - undefined, three.js - 97
mdns_discovery.js:55 Found HTTP service Object
port_handler.js:210 PortHandler - Found: [{"path":"COM4","displayName":"Betaflight STM32F405","vendorId":1155,"productId":22336}]
serial_backend.js:90 Connecting to: COM4
common.js:43 Chrome API Error: Failed to connect to the port..
 Traced Error
    at checkChromeRuntimeError (chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/common.js:1156:90)
    at chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/js/main.js:2174:27
checkChromeRuntimeError @ common.js:43
(anonymous) @ serial.js:52
serial.js:155 serial: failed to open serial port

Bwsbmw, 2023-08-22:
When connected, under device manager its showing up as just a USB Device on Com 4 on windows 11. Not "STM Virtual Com port like it does on the windows 10 machine. Do i need to install STM32 virtual com port Driver?

Bwsbmw, 2023-08-22:
installed STM32 Virtual COM port. now in device manager its showing up as STM Virtual Com Port (com4), no longer USB Serial Device (Com4) it still does not connect. i get the same Chrome API Error: Failed to connect to the port.., Failed to open port.

Bwsbmw, 2023-08-22:
OK UPDATE. After installing STM Virtual Com Port Driver, then running ImpulseRC Fixer i am now able to Connect to the FC's Manually like before. Although now they will also Go into DFU when prompted threw GUI, which they wouldnt before. upon a save in the CLI it still gets the Chrome API Error: but atleast i can get into DFU. I think I need to update Zadig. i have a screenshot of where i am at. if you could let me know if i need to update with Zadig, please let me know. i think thats the last thing from keeping the GUI from auto connecting, and Failing to connect to port. Thankyou for your time. its greatly appreciated. here is the Screenshot of where i am at. 
[image]

closed 2023-08-22 by Asizon

Pinnh, 2025-08-27:
In ubuntu，It's permission problem( configurator has no write permission), should open with super user