betaflight/betaflight-configurator #3571: Betaflight Configurator - confusing explaination for DShot/Dynamic Idle setting

jawz101, 2023-09-08:
### Describe the bug

[image]

### To Reproduce

Under PID Tuning tab, found when hovering over the Dynamic Idle help button

### Expected behavior

A consistent use of either the term DShot Idle or Dynamic Idle if they mean the same thing.  And if the current version of the configurator is for firmware 4.4.x then, according to the pop-up, this setting is no longer relevant- implies that it no longer does anything- then the setting should be removed from the ui altogether.

### Configurator version

10.9.0

McGiverGim, 2023-09-08:
No, it's not. Dshot idle and dynamic idle are the two different ways to set up the speed of the motors when idle. They are not the same.

nerdCopter, 2023-09-08:
@jawz101 , DShot Idle is this: 
[image]

jawz101, 2023-09-08:
That looks like it's called motor idle

nerdCopter, 2023-09-09:
> That looks like it's called motor idle

it's called "motor" idle, because when you use multishot, it's still an idle. when you use dshot then it's digital (dshot)

jawz101, 2023-09-09:
So Dynamic Idle is different than DShot Idle - which is a motor idle.  Motor idle can be either Dshot Idle or Multishot Idle.

How about replacing DShot Idle with the term 'Motor Idle' within the Dynamic Idle pop-up so the user can knows to which setting the pop-up description is referring?

haslinghuis, 2023-09-09:
Static or fixed idle ? @ctzsnooze

nerdCopter, 2023-09-09:
`Static motor idle` fine by me, but no one ever confused in the past :shrug:

haslinghuis, 2023-09-09:
[image]

Motor idle (%, static) should be hidden when Bidirectional Dshot is enabled.

jawz101, 2023-09-09:
This is all confusing.  Why can't choices which do not function be hidden if they are not being used?

PR #3573 (2023-09-16): Simplify motor idle config
Fixes #3571
closed 2023-09-16 by haslinghuis