betaflight/betaflight-configurator #3572: Enabling GPS with MSP protocol is confusing

ws998116, 2023-09-08:
### Describe the bug

**Using nightly version (10.10.0-debug-ff4b62d) of configurator!**

To enable GPS with MSP protocol, it takes the following steps.

1. Configure port as MSP. Save and Reboot
2. Enable GPS in Configuration Tab, go to GPS tab and switch protocol to MSP. Save and Reboot.
3. Enable GPS in Configuration Tab again. Save and Reboot.

If you do not switch the protocol before rebooting the second time, you will not be able to enable GPS because there is not a serial port configured for GPS.

### To Reproduce

1. Configure port as MSP. Save and Reboot
2. Enable GPS in Configuration Tab. Save and Reboot.
3. GPS tab is not available.

### Expected behavior

1. Configure port as MSP. Save and Reboot
2. Enable GPS with MSP protocol in Configuration Tab. Save and Reboot.
3. View data in GPS tab.

### Configurator version

10.10.0-debug-ff4b62d

### Add any other context about the problem that you think might be relevant here

Testing with HOBBYWING_XROTORF7CONV

haslinghuis, 2023-09-09:
Did you try to enable MSP and GPS at the same time (as it should enable feature GPS automatically)?
Which GPS is being used?

ws998116, 2023-09-11:
Enabling a port as MSP does not seem to enable GPS automatically. I cannot enable both at the same time because if I leave the `Ports` tab then the settings are not saved. Configuring the GPS protocol as MSP does not automatically enable GPS.

I am using the Flite Test EZ ID (Remote ID Broadcast Module) connected over UART 1.

The required settings are as follows. It is possible to configure it this way, but it is difficult to do because the GPS protocol configuration is on a different tab than enabling GPS.

[image]

[image]

[image]

haslinghuis, 2023-09-11:
- Enable MSP and GPS at the same time (as it should enable feature GPS automatically) on ports tab.
- Notice GPS is automatically enabled on configuration tab.
- Select MSP on GPS tab. We could look into eliminating this last step.

nerdCopter, 2023-09-11:
> * Enable MSP and GPS at the same time [...]

what is the requirement for this? on `master`, if i enable both (ports tab uart3), save&reboot. neither save. (i can save one or the other, but not both)

ws998116, 2023-09-11:
Sorry @haslinghuis, I still cannot replicate your steps. I have the same issue as @nerdCopter.

[image]

haslinghuis, 2023-09-11:
Unless I need a MSP enabled GPS I'm not able to save MSP on the ports tab UART associated with GPS.
Cannot test if GPS works with MSP while just choosing the protocol used on GPS tab.

Prajwal-Koirala, 2023-12-27:
Sure, here is a possible response for the github issue:

> Thanks for reporting this issue! I understand that you're having trouble enabling GPS with MSP protocol on Betaflight configurator. It seems like the current configuration workflow makes it difficult to enable both features simultaneously.

> I've seen other reports of this issue as well, so it's definitely something we need to address. I'm working on a fix that will allow you to enable both MSP and GPS on the same tab and save the settings together.

> In the meantime, as a workaround, you can try enabling MSP first, then switching to the Ports tab and enabling GPS there. This should allow you to use both features together.

> I'll keep you updated on the progress of the fix. Thanks for your patience!

Here are some additional things you may want to consider adding to your response:

* A link to the relevant documentation on enabling GPS and MSP in Betaflight configurator
* An offer to help the user troubleshoot their specific setup
* A request for more information from the user, such as their flight controller model and Betaflight version

I hope this helps!

ctzsnooze, 2024-01-05:
It is important to *never* enable GPS on a serial port if you are connecting your GPS over MSP.

AZDane, 2024-05-27:
> @ws998116 Can I consider the directions at the very beginning of this post to be a useable workaround to get GPS working over MSP? I'm struggling with the same issue with a Phoenix remote ID GPS passthrough that uses MSP and I can't bear to put more time into it unless I'm confident it'll work. No worries if it's not a workaround, any clarification is great. Thanks!
> 
> > To enable GPS with MSP protocol, it takes the following steps.
> > Configure port as MSP. Save and Reboot
> > Enable GPS in Configuration Tab, go to GPS tab and switch protocol to MSP. Save and Reboot.
> > Enable GPS in Configuration Tab again. Save and Reboot.
> > If you do not switch the protocol before rebooting the second time, you will not be able to enable GPS because there is not a serial port configured for GPS.

I had to do something to that extent when I set up my Phoenix remote ID module. Make sure to Save a Backup before you begin because if you end up doing it in the wrong order, I have seen it clear all your ports. During testing I have switched a lot between directly connected GPS/UBLOX and connected via Phoenixuas/MSP and because it is a little buggy, I just restore backups since its easier. Everytime you change, you also have to reenable GPS rescue under failsafe and reenable GPS Rescue switch in modes etc. 
Also, if you plan on using GPS Rescue over MSP make sure you are on 4.5.1 Dev since it has a bugfix for this.

kensherman, 2025-02-13:
I believe the work around is to after configuring your port, and enabling gps in the configuration and having both saved, go to the cli and type "set gps_provider = MSP" and save.

haslinghuis, 2025-02-14:
Some points for improvement:
- always show GPS tab when included in build options (perhaps only for expert mode)
- move GPS feature to GPS tab
- still need to enable MSP on UART used for GNSS in ports tab
- to revert to UBX we first need to disable GPS feature otherwise firmware locks up

PR #4344 (2025-02-16): Improve configuration for MSP GNSS
- resolves #3572
- enable gps tab if included in build options
- moves gps feature to gps tab

[image]
closed 2025-02-16 by haslinghuis