betaflight/betaflight-configurator #3592: OSD view is too large in 10.9

bbostian, 2023-10-01:
### Describe the bug

Version 10.9 the OSD tab the preview is too large.  This is simular to the issue [https://github.com/betaflight/betaflight-configurator/issues/3305](url) but reading the fix, that seemed to address MacOS.  My issue is on WIndows 10.

Betaflight 10.9 works fine on my laptop that has a screen resolution of 1920 x 1080 but on my field laptop, with a resultion of 1366 x 768, it has the issue.

This is 10.8 on the laptop with resolution of 1366 x 768:
[image]

Here is 10.9
[image]

### To Reproduce

I can only assumer you need a WIndows 10 laptop that has a resolution less than 1920 x 1080.

### Expected behavior

10.9 should display like 10.8.

### Configurator version

10.9.0

alanJames00, 2023-10-01:
Hello, @bbostian , I would love to work on this issue under hacktoberfest 2023. Would you please consider assigning this issue to me.

nerdCopter, 2023-10-02:
10.9 and now 10.10-RC must account for HD.
The problem arises from this plus the need for Android compatibility.
While i'm certain it could be fixed, I've personally tried various values in the `src/css/osd.less` file without acceptable results.  I can allow for a smaller preview, but doing so arranges the column 1 and column 3 in differing spots, whereas it is desired to both move to the bottom at once. (search for `osd-preview` and modify nearby values if anyone wishes to experiment)
An alternative might be to rearrange columns :shrug: .

closed 2023-11-09 by github-actions[bot]