betaflight/betaflight-configurator #3596: Firmware list unavailable in betaflight-configurator_10.9.0_win64-portable

swilkens, 2023-10-05:
### Describe the bug

When using betaflight-configurator_10.9.0_win64-portable (release), the firmware flasher appears stuck on "loading".

[image]

Current betaflight-configurator_10.10.0-RC1_win64-portable is working fine.

### To Reproduce

Download betaflight-configurator_10.9.0_win64-portable, unpack and start. Enter the firmware flasher.

### Expected behavior

Show a list of boards and firmware versions.

### Configurator version

10.9.0

### Flight controller configuration

```txt
n/a
```

nerdCopter, 2023-10-05:
not having the same issue here. maybe internet issue?
https://build.betaflight.com/api/targets online for me.

nerdCopter, 2023-10-05:
very odd. maybe some system npm_module taking orecedence :shrug: . i did test the portable; however, mine is linux.

swilkens, 2023-10-05:
Ran a folder compare between portable and installed windows versions, quite some differences found.

[image]

For now I'll just used the installed version.

nerdCopter, 2023-10-10:
concur.

Recommend not using any "shortcuts" as those may be pointing to another installation.  Use only the .exe when using portable versions. Otherwise, maybe delete everything, find all remnants to purge, then try again.

closing this ticket.

closed 2023-10-10 by nerdCopter

swilkens, 2023-10-10:
@nerdCopter I did some more checking.

Windows 10 PC - Portable not OK, installed version OK
Windows 11 Laptop - Portable OK, installed version OK

I can't explain it. maybe there's some issue I can't find on my PC. but I think I agree that the problem is somehow with me.