betaflight/betaflight-configurator #3633: Video Transmitter: wrong list of power levels for AKK Dominator VTX

gavrilow, 2023-11-14:
### Describe the bug

When configure VTX expect levels like 250, 500, 1000, 2000 mW levels in power field
[image]

### To Reproduce

1. Install and run betaflight-configurator_10.10.0-RC1_macOS
2. Connect drone with AKK Dominator VTX  to Betaflight
3. Move to Video Transmitter section
4. Open power list
5. Pay attention at power list 250, 500, 100, 200 mW (not 500, 1000, 2000, 2500...)

### Expected behavior

When configure VTX expect levels like 250, 500, 1000, 2000 mW levels in power field

### Configurator version

betaflight-configurator_10.10.0-RC1_macOS

McGiverGim, 2023-11-14:
If I'm not wrong, the power list is taken from the VTX Table, that is just at the bottom of this page.

McGiverGim, 2023-11-14:
One thing more about this: the power labels, are restricted to length 3. If you want to show values like 1000 or 2000, usually is better to put 1W or 2W (the label accepts text, not only numbers).

gavrilow, 2023-11-14:
> One thing more about this: the power labels, are restricted to length 3. If you want to show values like 1000 or 2000, usually is better to put 1W or 2W (the label accepts text, not only numbers).

So we need a field with 4 symbols. Not 3

McGiverGim, 2023-11-14:
No, we don't. As I say you can use letters to specify the values. If you "need" to show more digits, you can open a Pull Request (or a feature request) in the firmware part of Betaflight, that is where the limit is.

gavrilow, 2023-11-14:
How about 2.5W for example or 2500?

McGiverGim, 2023-11-14:
It's not perfect, but you can publish as:

```
.25
.50
1.0
2.5
```

To get more, as I say, the code needs to be modified in the firmware part (is not a limitation of the Configurator).

closed 2023-11-14 by McGiverGim