betaflight/betaflight-configurator #3649: HIFIONRC F7 only gets into DFU mode on 10.7.2 and nothing newer

slowthisbirddown, 2023-11-29:
### Describe the bug

DFU mode does not show up for this board on newer releases of BF. So I cannot flash the latest firmware for this board.
https://www.getfpv.com/hifionrc-f7-2-6s-25a-aio-fc-esc-toothpick-flight-controller.html

### To Reproduce

Boot into DFU mode. Nothing shows.

### Expected behavior

DFU mode to flash firmware

### Configurator version

Latest version 10.9.0

slowthisbirddown, 2023-11-29:
:( please help

nerdCopter, 2023-11-29:
Try [Configurator 10.10-rc1](https://github.com/betaflight/betaflight-configurator/releases/tag/10.10.0-RC1)  ?

slowthisbirddown, 2023-11-29:
> Try [Configurator 10.10-rc1](https://github.com/betaflight/betaflight-configurator/releases/tag/10.10.0-RC1) ?

I have and it's anything after the version listed in the title. No DFU on any newer version of Configurator. :(

nerdCopter, 2023-11-29:
Unless some dev has `HIFIONRC F7`, unsure how to proceed.
Please also specify your operating system and its build/version.

slowthisbirddown, 2023-11-29:
Windows 11 Pro 22H2 Windows Feature Experience Pack 1000.22677.1000.0

nerdCopter, 2023-11-29:
Recommend ImpulseRC driver fixer or other methods for completeness sake.
Maybe even running new Configurators in "compatibility mode".
This is all the notes i have regarding Windows USB problems: https://hackmd.io/@nerdCopter/rJv5TUrQ2

slowthisbirddown, 2023-11-29:
> ImpulseRC driver fixer

Tried the driver fixer. No avail. This is on a desktop and laptop. I don't know how to run it in compatibility mode. And I tried all the fixes on this list.

slowthisbirddown, 2023-11-29:
Nope. Nothing. This is insane. Never had an issue like this in all my time with betaflight.

haslinghuis, 2023-11-29:
Try Zadig

slowthisbirddown, 2023-11-29:
I had tried that as well.

nerdCopter, 2023-12-01:
solved @slowthisbirddown ??