betaflight/betaflight-configurator #3651: GPS not detected no matter what UART used. Solid lights on GPS.

slowthisbirddown, 2023-12-01:
### Describe the bug

Tried uarts 1,2 and 3 and RX to TX and TX to RX etc etc and enabled in ports menu and BF does not detect a GPS. This is with two models including one I received today (HGLRC M10 chip). Originally the GPS was working where it was originally plugged in. Have since full chip erase and flashed to latest 4.4.3 BF. GPS  in configuration does not remain flipped on and gets flipped off when saved.

### To Reproduce

Use any uart you wish and NO gps gets seen

### Expected behavior

GPS light to turn on

### Configurator version

Latest version 10.9.0

### Flight controller configuration

```txt
HIFIONRC F7
```

slowthisbirddown, 2023-12-01:
Okay red light is blinking on GPS unit. But everything remains the same.

nerdCopter, 2023-12-01:
Discord for "support" please. https://discord.betaflight.com/

closed 2023-12-01 by slowthisbirddown