betaflight/betaflight-configurator #3691: Save Button on LED Tab not working

TheIsotopes, 2023-12-23:
### Describe the bug

Save button on LED Tab not working

screenshot console output:
[image]

### To Reproduce

- go to led tab
- configure led's
- press save button
- go on another tab
- go back to led tab
- config was not saved

### Expected behavior

working save button

### Configurator version

last nightly

haslinghuis, 2023-12-23:
@TheIsotopes thanks for testing - much appreciated.

TheIsotopes, 2023-12-23:
@haslinghuis your welcome ;-)

#3692 tested and working

PR #3692 (2023-12-24): Revert "Add MSP support for led profile and status (#3671)"
Fixes #3691

This reverts commit b8da48af95c12d47e864529642a3e5a8e2baaf31so let's roll this to next release.
closed 2023-12-24 by haslinghuis