betaflight/betaflight-configurator #3707: (NEW TO FPV) 10.9.0 (67943b) is older than your frimware 4.5.0

UOdrones, 2023-12-30:
### Describe the bug

10.9.0 (67943b) is older than your firmware 4.5.0, is there a place to get this download? New to betaflight and cant get things working due to this.

### To Reproduce

after I connect to drone on betaflight

### Expected behavior

expected to be able to update frimware

### Support ID

```txt
2023-12-26 @20:21:52 -- Flight controller info, identifier: BTFL, version: 4.5.0

2023-12-26 @20:21:52 -- Running firmware released on: Dec 27 2023 01:07:02

2023-12-26 @20:21:52 -- Board: LMNR/LUXHDAIO-G4(STM32G47X), version: 0

2023-12-26 @20:21:52 -- Unique device ID: 0x4000284146501420303734

2023-12-26 @20:21:52 -- Craft name:

2023-12-26 @20:21:52 -- Arming Disabled
```

### Flight controller

DJI

### Other components

https://www.getfpv.com/fpv-quad-kits/lumenier-qav-pro-nano-whoop-2-cinequads-edition-rtf-bundle.html

Bought this bundle and was delivered 3 weeks ago. Cant get anything to work.

### How are the different components wired up (including port information)

Everything is connected from GETFPV.com everything talks to each other and is synced I just cant get the correct settings to load.

### Add any other context about the problem that you think might be relevant here

no idea

nerdCopter, 2023-12-31:
if not already done, maybe need to block @Prajwal-Koirala as seemingly many AI generated summaries, etc. are practically useless.

Prajwal-Koirala, 2023-12-31:
> if not already done, maybe need to block @Prajwal-Koirala as seemingly many AI generated summaries, etc. are practically useless.

Not really its a really good overview of the issue going so its easier for everyone else to understand the problem and not just you.

sugaarK, 2023-12-31:
> if not already done, maybe need to block @Prajwal-Koirala as seemingly many AI generated summaries, etc. are practically useless.

@blckmn

haslinghuis, 2024-01-01:
@Prajwal-Koirala 

Respecting your intentions but you could have asked our team if we are interested in the AI analysis.
For now please stop spamming our repo as I found some AI suggestions not in line with reality and clogging up my workflow.

closed 2024-01-31 by nerdCopter