betaflight/betaflight-configurator #3717: re-implement craft name on file output

pitts-mo, 2023-12-31:
### Describe the bug

10.10.0-rc1 appears to be missing craft name on file output. 

Craft name was recently re-implemented with PR #3637 and I still see the changes in master so I am not sure where the 10.10.0-rc1 release build failed to include this from master.

### To Reproduce

install 10.10.0-rc1 from official release assets and attempt to save a diff all from an FC with craft name.

BTFL_cli_20231231_100907_BETAFLIGHTF4.txt

(betaflight-configurator_10.10.0-RC1_amd64.deb)

### Expected behavior

craft name to be included in generated output file name somewhere.

eg:
BTFL_cli_20231231_100907_BETAFLIGHTF4_*craftname*.txt

### Configurator version

10.10.0-rc1

### Flight controller configuration

```txt
BTFL_cli_20231231_100907_BETAFLIGHTF4.txt
```

haslinghuis, 2023-12-31:
Fixed in nightly build 

[image]

haslinghuis, 2023-12-31:
Please check nightly build 

[image]

pitts-mo, 2023-12-31:
I am sorry. I grabbed a new configurator to go with 4.5.0-rc1 testing and didn't realize how old it was.
Thank you,
-p

closed 2023-12-31 by pitts-mo