betaflight/betaflight-configurator #3727: error in configuring the osd

zambon405, 2024-01-03:
### Describe the bug

I'm having difficulty configuring the OSD after the firmware update, I write the code in the image but it's giving an error, I don't have much knowledge in this part of programming.

### To Reproduce

[image]

### Expected behavior

I hope you can show the OSD information on the FPV goggles v2

### Configurator version

4.4.3

### Flight controller configuration

```txt
# diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.3 Nov 15 2023 / 11:43:33 (738127e7e) MSP API: 1.45

# config: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name IFLIGHT_BLITZ_F722_X1
manufacturer_id IFRC
mcu_id 004d002c3132511430393738
signature 

# name: Defender 25

# feature
feature -TELEMETRY
feature -AIRMODE
feature MOTOR_STOP
feature GPS
feature LED_STRIP

# serial
serial 0 1 115200 57600 0 115200
serial 2 2 115200 115200 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 1 1400 2100 0 0
aux 1 1 0 900 1275 0 0
aux 2 27 2 1300 1700 0 0
aux 3 28 1 1700 2100 0 0
aux 4 35 0 1700 2100 0 0
aux 5 40 7 900 2100 0 0

# master
set gyro_lpf1_static_hz = 337
set gyro_lpf2_static_hz = 0
set dyn_notch_count = 1
set gyro_lpf1_dyn_min_hz = 337
set gyro_lpf1_dyn_max_hz = 675
set acc_calibration = 14,42,-35,1
set rc_smoothing_setpoint_cutoff = 12
set rc_smoothing_feedforward_cutoff = 12
set serialrx_provider = SBUS
set sbus_baud_fast = ON
set blackbox_sample_rate = 1/16
set dshot_idle_value = 1200
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set motor_poles = 12
set failsafe_switch_mode = STAGE2
set align_board_yaw = 90
set vbat_max_cell_voltage = 422
set small_angle = 180
set gps_sbas_mode = AUTO
set gps_ublox_use_galileo = ON
set gps_rescue_ascend_rate = 500
set gps_rescue_return_alt = 50
set gps_rescue_ground_speed = 2000
set gps_rescue_max_angle = 32
set gps_rescue_descent_dist = 200
set gps_rescue_throttle_max = 1600
set gps_rescue_throttle_hover = 1280
set gps_rescue_sanity_checks = RESCUE_SANITY_ON
set pid_process_denom = 2
set simplified_gyro_filter_multiplier = 135
set osd_warn_bitmask = 90111
set osd_cap_alarm = 900
set osd_vbat_pos = 2432
set osd_tim_1_pos = 2456
set osd_tim_2_pos = 2456
set osd_remaining_time_estimate_pos = 2393
set osd_flymode_pos = 2444
set osd_mah_drawn_pos = 2432
set osd_gps_speed_pos = 2208
set osd_home_dir_pos = 2267
set osd_home_dist_pos = 2233
set osd_compass_bar_pos = 2058
set osd_altitude_pos = 2240
set osd_warnings_pos = 14634
set osd_avg_cell_voltage_pos = 2432
set osd_disarmed_pos = 2315
set osd_displayport_device = MSP
set osd_canvas_width = 30
set osd_canvas_height = 13
set debug_mode = GYRO_SCALED
set gyro_1_sensor_align = CW0FLIP
set gyro_1_align_yaw = 0
set craft_name = Defender 25
set pilot_name = zambon

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 0
set dterm_lpf1_dyn_max_hz = 202
set dterm_lpf1_static_hz = 101
set dterm_lpf2_static_hz = 202
set vbat_sag_compensation = 100
set yaw_lowpass_hz = 0
set p_pitch = 101
set i_pitch = 100
set d_pitch = 72
set f_pitch = 113
set p_roll = 97
set i_roll = 95
set d_roll = 70
set f_roll = 109
set p_yaw = 97
set i_yaw = 95
set f_yaw = 109
set d_min_roll = 58
set d_min_pitch = 59
set thrust_linear = 25
set simplified_master_multiplier = 140
set simplified_i_gain = 55
set simplified_d_gain = 140
set simplified_pi_gain = 155
set simplified_dmax_gain = 60
set simplified_feedforward_gain = 65
set simplified_pitch_d_gain = 90
set simplified_dterm_filter_multiplier = 135
set tpa_breakpoint = 1200

profile 1

profile 2

profile 3

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_srate = 100
set pitch_srate = 100
set yaw_srate = 70

rateprofile 1

rateprofile 2

rateprofile 3

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```

haslinghuis, 2024-01-03:
Should be fixed in #3720

haslinghuis, 2024-01-03:
For now use `set osd_displayport_device = MSP` instead.

Canaill51, 2024-01-03:
> For now use `set osd_displayport_device = MSP` instead.

Or also, just use "OSD for Fpv.wtf, DJI O3, Avatar HD' VTX preset? 
No CLI needs thanks to your job, BF Dev ;) Because we are all lazy shitty pilots lol

closed 2024-01-04 by haslinghuis