betaflight/betaflight-configurator #3743: OSD tab cannot be accessed after applying older configuration in client.

dafunkyyy, 2024-01-08:
### Describe the bug

The OSD tab cannot be selected in BF configurator 10.10 RC2

### To Reproduce

1- flash BF 4.5 RC2 with CrazybeeF4sx1280 with default cloud selection
2- go to client, paste old configuration from from BF 4.4.2
3- the OSD tab disappears
[image]

### Expected behavior

Should be able to see and select the OSD tab (image is taken before pasting old configuratin in CLI)
[image]

### Configurator version

10.10.0 RC2

### Flight controller configuration

```txt
# 

# diff

# version
# Betaflight / STM32F411 (S411) 4.5.0 Jan  8 2024 / 07:46:16 (f1cbd83f0) MSP API: 1.46
# config rev: ed9a9aa

# start the command batch
batch start

board_name CRAZYBEEF4SX1280
manufacturer_id HAMO

# name: Mobula6

# resources
resource MOTOR 1 B08
resource MOTOR 2 B07
resource MOTOR 3 B06
resource MOTOR 4 B10

# feature
feature -AIRMODE

# serial
serial 1 2048 115200 57600 0 115200

# map
map TAER1234

# aux
aux 0 0 0 1800 2100 0 0
aux 1 2 1 900 1100 0 0
aux 2 28 1 900 2100 0 0
aux 3 35 2 1800 2100 0 0
aux 4 47 1 1900 2100 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 LOWRACE  L FACTORY 5333 5373 5413 5453 5493 5533 5573 5613
vtxtable powerlevels 5
vtxtable powervalues 10 2 14 20 26
vtxtable powerlabels 0 RCE 25 100 400

# master
set gyro_lpf1_static_hz = 150
set gyro_lpf2_static_hz = 300
set dyn_notch_count = 2
set dyn_notch_max_hz = 700
set gyro_lpf1_dyn_min_hz = 150
set gyro_lpf1_dyn_max_hz = 300
set acc_lpf_hz = 10
set acc_calibration = -28,41,46,1
set rc_smoothing_auto_factor = 120
set blackbox_device = NONE
set dshot_bidir = ON
set motor_poles = 12
set align_board_yaw = 90
set bat_capacity = 300
set vbat_max_cell_voltage = 440
set vbat_min_cell_voltage = 320
set vbat_warning_cell_voltage = 340
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 2
set simplified_gyro_filter_multiplier = 60
set report_cell_voltage = ON
set osd_warn_bitmask = 24575
set osd_rssi_pos = 314
set osd_link_quality_pos = 2092
set osd_rssi_dbm_pos = 2060
set osd_tim_2_pos = 2455
set osd_flymode_pos = 409
set osd_throttle_pos = 2241
set osd_vtx_channel_pos = 2101
set osd_crosshairs_pos = 2317
set osd_current_pos = 2081
set osd_mah_drawn_pos = 2443
set osd_craft_name_pos = 394
set osd_warnings_pos = 2345
set osd_avg_cell_voltage_pos = 2433
set osd_disarmed_pos = 266
set osd_core_temp_pos = 34
set osd_canvas_height = 13
set vtx_band = 5
set vtx_channel = 1
set vtx_power = 5
set vtx_low_power_disarm = ON
set vtx_freq = 5658
set vcd_video_system = NTSC
set gyro_1_sensor_align = CW180
set gyro_1_align_yaw = 1800
set expresslrs_uid = 124,115,94,239,241,47
set rpm_filter_harmonics = 2
set craft_name = Mobula6

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 90
set dterm_lpf1_dyn_max_hz = 180
set dterm_lpf1_static_hz = 90
set dterm_lpf2_static_hz = 180
set vbat_sag_compensation = 100
set anti_gravity_gain = 90
set iterm_relax_cutoff = 5
set pidsum_limit = 1000
set pidsum_limit_yaw = 1000
set yaw_lowpass_hz = 125
set p_pitch = 75
set i_pitch = 134
set d_pitch = 76
set f_pitch = 199
set p_roll = 71
set i_roll = 127
set d_roll = 67
set f_roll = 191
set p_yaw = 71
set i_yaw = 127
set f_yaw = 191
set d_min_roll = 67
set d_min_pitch = 76
set thrust_linear = 20
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 65
set dyn_idle_min_rpm = 50
set simplified_master_multiplier = 160
set simplified_d_gain = 140
set simplified_dmax_gain = 0
set simplified_dterm_filter_multiplier = 120

rateprofile 0

# rateprofile 0
set thr_mid = 45
set thr_expo = 50
set roll_rc_rate = 1
set pitch_rc_rate = 1
set yaw_rc_rate = 1
set roll_expo = 55
set pitch_expo = 55
set yaw_expo = 45
set roll_srate = 72
set pitch_srate = 72

# end the command batch
batch end

#
```

### Add any other context about the problem that you think might be relevant here

Also the OSD doesn't display anything on the goggles except the warning

nerdCopter, 2024-01-08:
OSD enabled?
[image]

dafunkyyy, 2024-01-08:
You naited it !
ok so the OSD is enabled by default after flashing BF 4.5, however for some reason using the command 'defaults nosave' will disable the OSD.

I don't think it's supposed to do this if we want users to be able to take a 'diff all' from 4.3 and migrate to 4.5

Anyways Thanks, I can now test 4.5 !

haslinghuis, 2024-01-08:
The diff does not contain `feature OSD`

dafunkyyy, 2024-01-09:
no it does not, but using this diff in  4.2 or 4.3 will not disable OSD. Why would it disable it in 4.5 ?
I don't know many pilots who would want OSD disabled.

closed 2024-01-09 by dafunkyyy

nerdCopter, 2024-01-09:
@haslinghuis , is `#define DEFAULT_FEATURES (FEATURE_OSD)` valid in config.h's?  i only see 6 out of 436 config.h's with `DEFAULT_FEATURES`.

haslinghuis, 2024-01-09:
Mostly cloud driven, have to investigate why importing a diff disables the feature.

```
src/main/config/config.c:459:    featureDisableImmediate(FEATURE_OSD);
src/main/fc/init.c:945:          featureDisableImmediate(FEATURE_OSD);
src/main/config/config.c:491:    featureEnableImmediate(FEATURE_OSD);
```