betaflight/betaflight-configurator #3762: Unable to get GPS satellite lock.

Spit29, 2024-01-16:
### Describe the bug

Multiple satellites will keep appearing but none of them will lock. 

I'm using betaflight 4.5 RC2
GEPRC GEP-M10-DQ GPS Module

[image]

[image]

### To Reproduce

Normal GPS setup routine followed. I'm not sure if it's a bad GPS chip or a bug

### Expected behavior

I expected the satellites to lock and give me 3D fix

### Configurator version

10.10.0 RC2

### Flight controller configuration

```txt
# 

# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.5.0 Jan  9 2024 / 09:15:55 (f1cbd83f0) MSP API: 1.46
# config rev: 7982a5b

# start the command batch
batch start

board_name SPEEDYBEEF7V3
manufacturer_id SPBE

# feature
feature GPS

# serial
serial 0 131073 115200 57600 0 115200
serial 5 2 115200 115200 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 3 1700 2100 0 0
aux 2 13 1 1700 2100 0 0
aux 3 26 0 1725 2100 0 0
aux 4 35 2 1700 2100 0 0

# master
set gyro_lpf1_static_hz = 0
set gyro_lpf2_static_hz = 850
set dyn_notch_count = 4
set dyn_notch_min_hz = 30
set dyn_notch_max_hz = 400
set gyro_lpf1_dyn_min_hz = 0
set gyro_lpf1_dyn_max_hz = 875
set acc_lpf_hz = 10
set acc_calibration = 19,-13,49,1
set align_mag = CW0FLIP
set mag_align_pitch = 1800
set mag_hardware = NONE
set mag_calibration = 276,211,460
set rc_smoothing_auto_factor = 80
set rc_smoothing_setpoint_cutoff = 25
set rc_smoothing_feedforward_cutoff = 25
set blackbox_sample_rate = 1/1
set dshot_idle_value = 700
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_output_reordering = 2,3,0,1,4,5,6,7
set bat_capacity = 2200
set small_angle = 180
set gps_auto_config = OFF
set simplified_gyro_filter_multiplier = 170
set osd_units = IMPERIAL
set osd_alt_alarm = 350
set osd_link_quality_pos = 3393
set osd_rssi_dbm_pos = 3361
set osd_tim_2_pos = 3337
set osd_throttle_pos = 2107
set osd_current_pos = 2446
set osd_mah_drawn_pos = 2414
set osd_gps_sats_pos = 2065
set osd_home_dir_pos = 2106
set osd_compass_bar_pos = 72
set osd_altitude_pos = 19490
set osd_warnings_pos = 14518
set osd_avg_cell_voltage_pos = 2384
set osd_battery_usage_pos = 2069
set osd_displayport_device = MSP
set osd_canvas_width = 60
set osd_canvas_height = 22
set debug_mode = GYRO_SCALED
set vcd_video_system = HD
set gyro_2_sensor_align = CW180
set gyro_2_align_pitch = 0
set gyro_2_align_yaw = 1800

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 0
set dterm_lpf1_dyn_max_hz = 157
set dterm_lpf1_type = PT2
set dterm_lpf1_static_hz = 112
set dterm_lpf2_static_hz = 0
set iterm_relax = RPY
set iterm_relax_cutoff = 5
set p_pitch = 70
set i_pitch = 69
set d_pitch = 60
set f_pitch = 149
set p_roll = 58
set i_roll = 57
set d_roll = 46
set f_roll = 124
set p_yaw = 58
set i_yaw = 57
set f_yaw = 124
set d_min_roll = 46
set d_min_pitch = 60
set feedforward_smooth_factor = 30
set feedforward_jitter_factor = 10
set simplified_master_multiplier = 130
set simplified_i_gain = 55
set simplified_d_gain = 120
set simplified_dmax_gain = 0
set simplified_feedforward_gain = 80
set simplified_pitch_d_gain = 115
set simplified_pitch_pi_gain = 115
set simplified_dterm_filter_multiplier = 150

rateprofile 0

# rateprofile 0
set roll_rc_rate = 1
set pitch_rc_rate = 1
set yaw_rc_rate = 6
set roll_expo = 55
set pitch_expo = 55
set yaw_expo = 35
set roll_srate = 60
set pitch_srate = 60
set yaw_srate = 75

# end the command batch
batch end

#
```

### Add any other context about the problem that you think might be relevant here

This chip has not been tested on any other device. It didn't work in 4.4.3 either.

Pancronos, 2024-02-01:
try to enable galileo

haslinghuis, 2024-02-01:
enable auto config

Spit29, 2024-02-02:
It ended up being a faulty GPS unit. I had it replaced and everything works now.

closed 2024-02-02 by Spit29