betaflight/betaflight-configurator #3764: notify you when long operations are completed

mituritsyn, 2024-01-16:
### Is your feature request related to a problem? Please describe

There are a number of operations that take a long time:
-clearing the flash memory
-downloading the log
-flashing
I would like to have a notification when such operations are completed, so that I can switch to another programme and not forget at the end.

### Describe the solution you'd like

perform colour notification of BFC icon

### Describe alternatives you've considered

one more way is to bring window on top but is not the best solution

mituritsyn, 2024-02-16:
> This issue has been automatically marked as stale because it has not had recent activity. It will be closed if no further activity occurs within a week.

go away)

turboMaCk, 2024-03-12:
This should be relatively easy to implement using electron's API it seems (see https://www.electronjs.org/docs/latest/tutorial/notifications)

I might look into it myself if no one else wants to.

McGiverGim, 2024-05-15:
We don't use electron, the next version will be PWA. So if you want to to look at it, look at PWA or maybe some Vue component.

haslinghuis, 2024-05-15:
Sound alert

YarosMallorca, 2024-08-14:
I'll look into this, would be neat

YarosMallorca, 2024-08-16:
@haslinghuis this should be closed.

nerdCopter, 2024-08-16:
resolved by https://github.com/betaflight/betaflight-configurator/pull/4134