betaflight/betaflight-configurator #3768: SkyStars H7 (stack) on 4.5 gives me a "motor_protocol"error

incarsonated, 2024-01-19:
### Describe the bug

Motor protocol error in 4.5 using dshot300; I disabled and rebooted, same error. reflashed 4.5, same error. Not until I reflashed down to 4.3 did I get the flag to go away. I did not try 4.4. Also tried dshot600 as well with no luck

### To Reproduce

Simply dl 4.5 with SS H7 fc, karate freestyle preset.

### Expected behavior

no flag

### Configurator version

10.10 and 4.5

### Flight controller configuration

```txt
# version
# Betaflight / STM32H743 (SH74) 4.5.0 Dec 29 2023 / 03:16:07 (22ce58655) MSP API: 1.46
# config rev: 18e0321

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name SKYSTARSH7HD
manufacturer_id SKST
mcu_id 003f002c3432510631313336
signature 

# name: APEX VSTA

# feature
feature OSD
feature ESC_SENSOR

# serial
serial 0 64 115200 57600 0 115200
serial 2 1024 115200 57600 0 115200
serial 5 131073 115200 57600 0 115200

# beacon
beacon RX_SET

# led
led 0 5,7::CIW:2
led 1 6,7::CIW:2
led 2 7,7::CIW:2
led 3 8,7::CIW:2

# aux
aux 0 0 0 1875 2100 0 0
aux 1 1 1 900 1100 0 0
aux 2 2 1 1400 1625 0 0
aux 3 13 2 1900 2100 0 0
aux 4 13 0 900 900 0 35
aux 5 35 3 1900 2100 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 3
vtxtable powervalues 0 1 2
vtxtable powerlabels 25 200 500

# master
set gyro_lpf1_static_hz = 500
set gyro_lpf2_static_hz = 1000
set dyn_notch_count = 2
set dyn_notch_q = 500
set dyn_notch_min_hz = 120
set gyro_lpf1_dyn_min_hz = 500
set gyro_lpf1_dyn_max_hz = 1000
set acc_lpf_hz = 10
set acc_calibration = -104,-35,5,1
set baro_hardware = NONE
set rc_smoothing_auto_factor = 45
set rc_smoothing_auto_factor_throttle = 45
set serialrx_provider = CRSF
set blackbox_sample_rate = 1/2
set dshot_idle_value = 450
set motor_pwm_protocol = DSHOT300
set vbat_min_cell_voltage = 320
set yaw_motors_reversed = ON
set small_angle = 180
set simplified_gyro_filter_multiplier = 200
set osd_units = IMPERIAL
set osd_warnings_pos = 14355
set osd_avg_cell_voltage_pos = 2442
set osd_canvas_width = 30
set osd_canvas_height = 13
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 2
set vtx_freq = 5917
set rpm_filter_fade_range_hz = 100
set craft_name = APEX VSTA
set pilot_name = SSTARS-H7/60A

profile 0

# profile 0
set profile_name = Karate
set dterm_lpf1_dyn_expo = 10
set p_pitch = 59
set i_pitch = 106
set d_pitch = 58
set f_pitch = 158
set p_roll = 51
set i_roll = 92
set d_roll = 53
set f_roll = 138
set horizon_level_strength = 50
set d_min_roll = 43
set d_min_pitch = 46
set feedforward_smooth_factor = 30
set feedforward_jitter_factor = 10
set dyn_idle_min_rpm = 35
set dyn_idle_p_gain = 35
set simplified_pids_mode = RP
set simplified_d_gain = 145
set simplified_pi_gain = 115
set simplified_dmax_gain = 70
set simplified_feedforward_gain = 115
set simplified_pitch_d_gain = 95
set simplified_pitch_pi_gain = 110

profile 1

# profile 1
set profile_name = Spatz
set dterm_lpf1_dyn_min_hz = 78
set dterm_lpf1_dyn_max_hz = 157
set dterm_lpf1_static_hz = 78
set dterm_lpf2_static_hz = 157
set pidsum_limit = 1000
set pidsum_limit_yaw = 1000
set yaw_lowpass_hz = 0
set p_pitch = 69
set i_pitch = 124
set d_pitch = 45
set f_pitch = 219
set p_roll = 66
set i_roll = 118
set f_roll = 210
set p_yaw = 66
set i_yaw = 118
set f_yaw = 210
set d_min_roll = 40
set d_min_pitch = 45
set feedforward_smooth_factor = 30
set feedforward_jitter_factor = 10
set dyn_idle_min_rpm = 35
set simplified_master_multiplier = 135
set simplified_pi_gain = 110
set simplified_dmax_gain = 0
set simplified_feedforward_gain = 130
set simplified_dterm_filter_multiplier = 105

profile 2

# profile 2
set profile_name = SupaFly
set dterm_lpf1_dyn_min_hz = 0
set dterm_lpf1_dyn_max_hz = 0
set dterm_lpf1_dyn_expo = 0
set dterm_lpf1_static_hz = 78
set dterm_lpf2_static_hz = 157
set anti_gravity_gain = 100
set p_pitch = 59
set i_pitch = 90
set d_pitch = 33
set f_pitch = 129
set p_roll = 54
set i_roll = 82
set d_roll = 29
set f_roll = 118
set p_yaw = 54
set i_yaw = 82
set f_yaw = 118
set horizon_level_strength = 50
set d_min_roll = 29
set d_min_pitch = 33
set feedforward_smooth_factor = 30
set simplified_master_multiplier = 110
set simplified_i_gain = 85
set simplified_d_gain = 90
set simplified_pi_gain = 110
set simplified_dmax_gain = 0
set simplified_feedforward_gain = 90
set simplified_pitch_pi_gain = 105
set simplified_dterm_filter_multiplier = 105

profile 3

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 3
set pitch_rc_rate = 3
set yaw_rc_rate = 3
set roll_expo = 68
set pitch_expo = 72
set yaw_expo = 64
set roll_srate = 70
set pitch_srate = 75
set yaw_srate = 65

rateprofile 1

rateprofile 2

rateprofile 3

# restore original rateprofile selection
rateprofile 0

# save configuration
```

### Add any other context about the problem that you think might be relevant here

happy to discuss over phone as well. 706.897.7799. Thank you for all you do!

haslinghuis, 2024-01-23:
To check have to make some steps.

Please try 4.5 zulu 
- on defaults first
- after enabling dshot bidirectional
- after preset applied

haslinghuis, 2024-02-20:
Closing as stale

closed 2024-02-20 by haslinghuis