betaflight/betaflight-configurator #3774: fall from the sky after updating to 4.5RC2

Babangdes, 2024-01-22:
### Describe the bug

https://github.com/betaflight/betaflight-configurator/assets/157426315/36ed2116-5d4f-432a-b1be-74762237fdec

I have been flying the RC1 Long Range for close to a month now and it works just like a charm. All my flights from since November has been done with that drone and From December 5th I think I have the RC1 and it was just perfect.
I tried to RC2 and the quad start flying then after 2minutes or i did a split S and motors are running perfectly but M2 only stay below 10%. then the quad went to a death roll. i change the RC2 to RC1 and the quad keep flying just as it should.

### To Reproduce

Im not sure how to reproduce it.

### Expected behavior

I went to do a dive and the drone start going nicely then suddenly it went to a death roll. 

### Configurator version

Betaflight 4.5 RC2

### Flight controller configuration

```txt
# 

# dump all

# version
# Betaflight / STM32F7X2 (S7X2) 4.5.0 Dec 25 2023 / 10:06:01 (22ce58655) MSP API: 1.46
# config rev: 9bf017a

# start the command batch
batch start

board_name IFLIGHT_BLITZ_F722
manufacturer_id IFRC
mcu_id 003200183432511036303838
signature 

# name: Helion

# resources
resource BEEPER 1 C13
resource MOTOR 1 B01
resource MOTOR 2 B00
resource MOTOR 3 C08
resource MOTOR 4 C09
resource MOTOR 5 NONE
resource MOTOR 6 B07
resource MOTOR 7 B10
resource MOTOR 8 B11
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 C10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 C11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
[... 1437 lines cut ...]
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 67
set pitch_srate = 67
set yaw_srate = 67
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#
```

### Add any other context about the problem that you think might be relevant here

how can i send the Blackbox to you?
I have a link here with the blackbok
[https://1drv.ms/f/s!Aq_anDyPBe6IhYkPrF_uoL0nJqbwDQ?e=G57WZB ](url)

haslinghuis, 2024-01-23:
Please supply support ID (you can find the button in CLI tab).

- Try zulu development build
- Try enabling dshot_bitbang

sugaarK, 2024-01-24:
got a black box file? looks like an esc issue

sugaarK, 2024-01-24:
Your link doesn’t work for me and if it was a common issue eg not just you we’d be seeing more of these…  800+ people a day remember flashing rc2. I want diff too dump is not very helpful for problem solving

Babangdes, 2024-01-24:
# 

# Diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.5.0 Dec 25 2023 / 10:06:01 (22ce58655) MSP API: 1.46
# config rev: 9bf017a

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name IFLIGHT_BLITZ_F722
manufacturer_id IFRC
mcu_id 003200183432511036303838
signature 

# name: Helion

# resources
resource MOTOR 5 NONE
resource PINIO 1 B06

# feature
feature GPS
feature TELEMETRY
feature LED_STRIP
feature OSD

# serial
serial 0 131073 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 3 2 115200 115200 0 115200

# led
led 0 0,0::C:8
led 1 1,0::C:8
led 2 2,0::C:8
led 3 3,0::C:8
led 4 4,0::C:8
led 5 5,0::C:8
led 6 6,0::C:8

# color
color 5 91,0,255
color 6 108,0,255

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 1700 2100 0 0
aux 2 46 3 1300 2100 0 0
aux 3 13 2 1300 1700 0 0
aux 4 35 2 1700 2100 0 0
aux 5 40 2 1600 2100 0 0
aux 6 40 1 1200 1700 0 0

# rxfail
rxfail 3 s 1350
rxfail 5 s 1800
rxfail 7 s 1800

# master
set gyro_lpf1_static_hz = 0
set gyro_lpf2_static_hz = 300
set dyn_notch_count = 2
set dyn_notch_q = 500
set dyn_notch_min_hz = 60
set gyro_lpf1_dyn_min_hz = 0
set gyro_lpf1_dyn_max_hz = 0
set acc_lpf_hz = 10
set acc_calibration = -34,-2,9,1
set rc_smoothing_setpoint_cutoff = 15
set rc_smoothing_feedforward_cutoff = 25
set blackbox_sample_rate = 1/2
set dshot_idle_value = 700
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set failsafe_delay = 4
set failsafe_procedure = GPS-RESCUE
set yaw_motors_reversed = ON
set small_angle = 180
set gps_sbas_mode = AUTO
set gps_ublox_use_galileo = ON
set gps_rescue_ground_speed = 1000
set gps_rescue_max_angle = 60
set gps_rescue_throttle_hover = 1300
set gps_rescue_allow_arming_without_fix = ON
set yaw_deadband = 5
set simplified_gyro_filter = OFF
set simplified_gyro_filter_multiplier = 105
set osd_rssi_dbm_alarm = -100
set osd_vbat_pos = 2304
set osd_rssi_pos = 3593
set osd_link_quality_pos = 3624
set osd_rssi_dbm_pos = 3561
set osd_tim_1_pos = 3119
set osd_tim_2_pos = 3151
set osd_flymode_pos = 2668
set osd_throttle_pos = 2487
set osd_vtx_channel_pos = 373
set osd_current_pos = 2368
set osd_mah_drawn_pos = 2400
set osd_craft_name_pos = 2048
set osd_gps_speed_pos = 3469
set osd_gps_lon_pos = 2240
set osd_gps_lat_pos = 2208
set osd_gps_sats_pos = 3337
set osd_home_dir_pos = 2552
set osd_home_dist_pos = 3405
set osd_flight_dist_pos = 3437
set osd_altitude_pos = 19757
set osd_power_pos = 512
set osd_warnings_pos = 14676
set osd_avg_cell_voltage_pos = 2336
set osd_disarmed_pos = 2677
set osd_core_temp_pos = 2432
set osd_gps_sats_show_hdop = ON
set osd_displayport_device = MSP
set osd_camera_frame_width = 30
set osd_camera_frame_height = 16
set debug_mode = GYRO_SCALED
set vcd_video_system = HD
set pinio_box = 40,41,42,43
set rpm_filter_weights = 100,20,100
set rpm_filter_min_hz = 50
set rpm_filter_fade_range_hz = 30
set craft_name = Helion
set pilot_name = Babang

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 0
set dterm_lpf1_dyn_max_hz = 157
set dterm_lpf1_type = PT2
set dterm_lpf1_static_hz = 63
set dterm_lpf2_static_hz = 0
set anti_gravity_gain = 90
set iterm_relax_cutoff = 8
set pidsum_limit = 1000
set pidsum_limit_yaw = 1000
set yaw_lowpass_hz = 50
set p_pitch = 93
set i_pitch = 66
set d_pitch = 114
set f_pitch = 148
set p_roll = 81
set i_roll = 57
set d_roll = 91
set f_roll = 129
set p_yaw = 81
set i_yaw = 57
set f_yaw = 129
set d_min_roll = 91
set d_min_pitch = 114
set thrust_linear = 10
set dyn_idle_min_rpm = 15
set simplified_master_multiplier = 180
set simplified_i_gain = 40
set simplified_d_gain = 170
set simplified_dmax_gain = 0
set simplified_feedforward_gain = 60
set simplified_pitch_d_gain = 110
set simplified_pitch_pi_gain = 110
set simplified_dterm_filter_multiplier = 85
set tpa_rate = 50
set tpa_breakpoint = 1200

profile 1

profile 2

profile 3

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 30
set roll_srate = 50
set pitch_srate = 50
set yaw_srate = 50

rateprofile 1

rateprofile 2

rateprofile 3

# restore original rateprofile selection
rateprofile 0

sugaarK, 2024-01-24:
@Babangdes  are you on our discord ? The bb link is Broken or locked for me. If your on our discord you can send me it to me there

nerdCopter, 2024-01-24:
i fixed the BBL link in original post

Babangdes, 2024-01-24:
@sugaarK the link works now.

sugaarK, 2024-01-24:
> @sugaarK the link works now.

thanks which log?

Babangdes, 2024-01-24:
btfl_003.bbl