betaflight/betaflight-configurator #3784: Radio protocol lost in 10.10.0 RC2

kmaxium, 2024-02-05:
### Describe the bug

[image]
[image]

### To Reproduce

don't know how, just report the issue.

### Expected behavior

Please fix it in the future release.

### Configurator version

10.10.2 rc2

nerdCopter, 2024-02-05:
One protocol only in the drop-down is by design.

if you want to **_add_** protocols, then do so in the custom defines field.

example: 
[image]

https://hackmd.io/@nerdCopter/H11rvS8Lh

closed 2024-02-05 by nerdCopter

kmaxium, 2024-02-25:
Got it, many thx