betaflight/betaflight-configurator #3785: GPS Configurations with Flite Test EZ ID

JohanTakeshi, 2024-02-06:
### Describe the bug

Betaflight and Flight Controller stop responding after choosing anything but "MSP" for GPS configuration.  Requires a reboot of Flight Controller.

### To Reproduce

2024-02-06 @10:30:46 -- Flight controller info, identifier: BTFL, version: 4.5.0
2024-02-06 @10:30:46 -- Running firmware released on: Jan 24 2024 18:35:06
2024-02-06 @10:30:46 -- Board: IFRC/IFLIGHT_SUCCEX_E_F7(STM32F7X2), version: 0
2024-02-06 @10:30:46 -- Unique device ID: 0x2200323756501720393558
2024-02-06 @10:30:46 -- Build Key: 36fb84a17309dcfbf51859d0d0a554e8

TX on Flite Test EZ ID (GPS) Connected to RX on Flight Controller, "MSP" Selected in "ports" tab.  If I change ports tab to "msp disabled" and "GPS" for "sensor input", then go to GPS and select "UBLOX" or "NMEA"...can't save the configuration and everything stops responding. 

### Expected behavior

It should either work, report an error or just NOT work and NOT FREEZE, I would think?

### Configurator version

10.10.0

### Flight controller configuration

```txt
# 

# diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.5.0 Jan 24 2024 / 18:35:06 (f1cbd83f0) MSP API: 1.46
# config rev: 1955e3c

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name IFLIGHT_SUCCEX_E_F7
manufacturer_id IFRC
mcu_id 002200323756501720393558
signature 

# name: ElRoachO

# resources
resource SERIAL_TX 4 NONE

# feature
feature -AIRMODE
feature INFLIGHT_ACC_CAL
feature SOFTSERIAL
feature GPS
feature TELEMETRY
feature LED_STRIP
feature OSD

# serial
serial 0 64 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 3 2048 115200 57600 0 115200
serial 4 2 115200 57600 0 115200

# beeper
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -DISARM_REPEAT
beeper -ARMED
beeper -ON_USB
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_LOST
beacon -RX_SET

# mode_color
mode_color 6 1 2

# aux
aux 0 0 0 1625 2025 0 0
aux 1 1 4 925 1225 0 0
aux 2 2 4 1325 1725 0 0
aux 3 27 5 1675 2075 0 0
aux 4 19 1 1300 1700 0 0
aux 5 28 3 1300 1700 0 0
aux 6 31 2 1700 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 14 23 27 29
vtxtable powerlabels 25 200 500 800

# vtx
vtx 0 7 0 0 1 950 1050
vtx 1 7 0 0 2 1150 1250
vtx 2 7 0 0 3 1350 1450
vtx 3 7 0 0 4 1550 1650

# master
set acc_calibration = 4,7,192,1
set min_check = 1000
set max_check = 2000
set fpv_mix_degrees = 30
set blackbox_sample_rate = 1/2
set min_throttle = 1069
set dshot_idle_value = 500
set dshot_bidir = ON
set motor_output_reordering = 2,3,0,1,4,5,6,7
set failsafe_off_delay = 20
set failsafe_throttle = 1164
set failsafe_throttle_low_delay = 10
set failsafe_procedure = AUTO-LAND
set failsafe_recovery_delay = 10
set bat_capacity = 1500
set vbat_min_cell_voltage = 320
set vbat_scale = 111
set ibata_scale = 16000
set ibata_offset = -1710
set yaw_motors_reversed = ON
set small_angle = 25
set gps_provider = MSP
set gps_ublox_use_galileo = ON
set gps_set_home_point_once = ON
set gps_rescue_alt_mode = FIXED_ALT
set gps_rescue_descent_dist = 30
set gps_rescue_throttle_min = 1000
set gps_rescue_throttle_max = 2000
set gps_rescue_throttle_hover = 1180
set gps_rescue_min_sats = 7
set gps_rescue_allow_arming_without_fix = ON
set runaway_takeoff_deactivate_throttle_percent = 10
set osd_units = IMPERIAL
set osd_warn_bitmask = 40959
set osd_rssi_dbm_alarm = -95
set osd_cap_alarm = 1000
set osd_alt_alarm = 20
set osd_vbat_pos = 2241
set osd_link_quality_pos = 2102
set osd_rssi_dbm_pos = 2134
set osd_flymode_pos = 2454
set osd_anti_gravity_pos = 2458
set osd_throttle_pos = 2200
set osd_vtx_channel_pos = 2420
set osd_crosshairs_pos = 2253
set osd_ah_sbar_pos = 2254
set osd_ah_pos = 2126
set osd_current_pos = 2304
set osd_mah_drawn_pos = 2272
set osd_craft_name_pos = 2049
set osd_gps_speed_pos = 2060
set osd_gps_lon_pos = 2407
set osd_gps_lat_pos = 2439
set osd_gps_sats_pos = 2403
set osd_home_dir_pos = 2147
set osd_home_dist_pos = 2177
set osd_flight_dist_pos = 2209
set osd_compass_bar_pos = 74
set osd_altitude_pos = 2091
set osd_esc_tmp_pos = 363
set osd_esc_rpm_pos = 2232
set osd_core_temp_pos = 2390
set osd_efficiency_pos = 2337
set osd_sys_goggle_voltage_pos = 2166
set osd_stat_bitmask = 216284908
set osd_canvas_width = 30
set osd_canvas_height = 13
set vtx_band = 1
set vtx_channel = 1
set vtx_power = 1
set vtx_freq = 5865
set displayport_msp_fonts = 0,0,0,0
set pinio_box = 0,255,255,43
set gyro_2_bustype = SPI
set gyro_2_sensor_align = CW0
set craft_name = ElRoachO

profile 0

# profile 0
set d_pitch = 37
set d_roll = 32
set horizon_level_strength = 50
set d_min_roll = 25
set d_min_pitch = 28
set simplified_d_gain = 85
set simplified_dmax_gain = 85

profile 1

# profile 1
set p_pitch = 51
set d_pitch = 42
set p_roll = 46
set d_roll = 39
set p_yaw = 50
set horizon_level_strength = 50
set d_min_roll = 25
set d_min_pitch = 28

profile 2

# profile 2
set horizon_level_strength = 50

profile 3

# profile 3
set horizon_level_strength = 50

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set thr_expo = 65
set roll_expo = 37
set pitch_expo = 36
set yaw_expo = 36

rateprofile 1

rateprofile 2

rateprofile 3

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#
```

### Add any other context about the problem that you think might be relevant here

I have also tried with "TX" and "RX" connected for Flite Test...same results. 
Have tried with the Flite Test module powered down completely - still results in "locked up" - Don't think it's a module issue causing the lock.

haslinghuis, 2024-02-06:
When GPS sensor is enabled on ports tab board also hangs when using CLI:

```
# master
set gps_provider = MSP
# set gps_provider = UBLOX
gps_provider set to UBLOX
```

Now have to use defaults to get it changed.

haslinghuis, 2024-02-06:
https://github.com/betaflight/betaflight/blob/24e7dabed3174d89db1f9f73a80d8649b40f57cb/src/main/config/config.c#L236-L241

Issue seems to be removing the port identifier for MSP. So we cannot not return.

Found a workaround: When switching from MSP - after removing GPS from ports tab - using CLI returning to UBLOX does not have this issue.