betaflight/betaflight-configurator #3791: Disable update modal for release candidates

sundowndev, 2024-02-10:
### Is your feature request related to a problem? Please describe

Release candidates are not like any other release and should be considered instable. Most users don't know what a release candidate is and will simply update the configurator if it tells them to do so. However it can lead to configuration issues and even lead to fly issues. It's dangerous to use it in the wild, especially when the user has no idea of the consequences.

### Describe the solution you'd like

Do not trigger a modal for release candidates. Instead, trigger it when the version is anterior to the latest stable version.

### Describe alternatives you've considered

N/A

### Other information

N/A

haslinghuis, 2024-02-11:
Did you check

[image]

haslinghuis, 2024-02-11:
Be aware sometimes RC release are needed for later RC firmware releases.

sundowndev, 2024-02-12:
@haslinghuis my bad, it does the job. Is it enabled by default?

sundowndev, 2024-02-15:
Thanks for the info. Closing.

closed 2024-02-15 by sundowndev