betaflight/betaflight-configurator #3796: FPort telemetry not working after cloud build

TheIsotopes, 2024-02-14:
### Describe the bug

since #3726 the fport telemetry is no longer working 

As you can see, the entry for the telemetry protocol is missing in the cloud build -> https://[build.betaflight.com/api/builds/78006759ad777eef8e55f2a2ddc32534/log](https://build.betaflight.com/api/builds/78006759ad777eef8e55f2a2ddc32534/log)

After entering the telemetry protocol in the custom defines line it works again -> https://[build.betaflight.com/api/builds/3de576f28d121f65183a9fa4ae399669/log](https://build.betaflight.com/api/builds/3de576f28d121f65183a9fa4ae399669/log)

There is no automatic setting of the telemetry protocol as it works with the ghost protocol.

@haslinghuis pls can you check this

### To Reproduce

see describe the bug

### Expected behavior

insert telemetry protocol for fport radio protocol in cloud build

### Configurator version

latest master

haslinghuis, 2024-02-14:
@blckmn cloud build is adding telemetry for CRSF and GHST as we don't add them in common*.h

Please add `USE_TELEMETRY_SMARTPORT` if `USE_SERIALRX_FPORT` is defined.

nerdCopter, 2024-02-16:
@blckmn , i realize that  comments were in different places. was this rectified in the backend yet?

blckmn, 2024-02-17:
This is now sorted.

closed 2024-02-17 by blckmn