betaflight/betaflight-configurator #3807: hi, I downloaded the betaflight-configurator_10.10.0-RC2 update forw11

BrownieIT, 2024-02-19:
### Describe the bug

hi, I downloaded the betaflight-configurator_10.10.0-RC2 update and when I click on install it doesn't start as if I didn't click

I'll start by saying that I only have problems with this version with zero problems with the previous ones, does anyone know how I can install it? I tried both the exe and the zip version

### To Reproduce

hi, I downloaded the betaflight-configurator_10.10.0-RC2 update and when I click on install it doesn't start as if I didn't click

I'll start by saying that I only have problems with this version with zero problems with the previous ones, does anyone know how I can install it? I tried both the exe and the zip version

### Expected behavior

hi, I downloaded the betaflight-configurator_10.10.0-RC2 update and when I click on install it doesn't start as if I didn't click

I'll start by saying that I only have problems with this version with zero problems with the previous ones, does anyone know how I can install it? I tried both the exe and the zip version

### Configurator version

betaflight-configurator_10.10.0-RC2

nerdCopter, 2024-02-20:
Possibly your download may be corrupt, or more likely your virus scanner sees it as a false-positive.  please white-list the installer and the application in your virus scanner, or try a nightly configurator.

haslinghuis, 2024-02-20:
Tried updating NWjs without a difference. Windows Defender has a problem with the icon creation.
On W10 I need to add an exclusion for the folder where configurator will be installed.

```
Using v0.83.0 (sdk)
Create cache folder in C:\dev\betaflight-configurator\cache\0.83.0-sdk
Using cache for: win64
Create release folder in C:\dev\betaflight-configurator\debug\betaflight-configurator\win64
Update win64 executable icon

Unhandled Exception: System.IO.DirectoryNotFoundException: Could not find a part of the path 'C:\Users\Mark\AppData\Local\Anolis\Exceptions.log'.
   at System.IO.__Error.WinIOError(Int32 errorCode, String maybeFullPath)
   at System.IO.FileStream.Init(String path, FileMode mode, FileAccess access, Int32 rights, Boolean useRights, FileShare share, Int32 bufferSize, FileOptions options, SECURITY_ATTRIBUTES secAttrs, String msgPath, Boolean bFromProxy)
   at System.IO.FileStream..ctor(String path, FileMode mode, FileAccess access, FileShare share, Int32 bufferSize, FileOptions options, String msgPath, Boolean bFromProxy)
   at System.IO.FileStream..ctor(String path, FileMode mode, FileAccess access, FileShare share, Int32 bufferSize, FileOptions options)
   at System.IO.StreamWriter..ctor(String path, Boolean append, Encoding encoding, Int32 bufferSize)
   at System.IO.StreamWriter..ctor(String path, Boolean append)
   at Anolis.Resourcer.Program.Main(String[] args)
Error building NW apps: Error while updating the Windows icon.
```

haslinghuis, 2024-04-14:
Latest nightly builds do install without problem on Windows 10 now as before I had to add an exception in Windows Defender.

closed 2024-05-05 by haslinghuis