betaflight/betaflight-configurator #3809: hmc5883l magnetometer

spinorose, 2024-02-20:
### Is your feature request related to a problem? Please describe

I went in the CLI to activate the compass and still cannot see it 

### Describe the solution you'd like

Can a prepopulated compass menu be made to make the whole experience easy and a details steps/documentation to activated .

### Describe alternatives you've considered

drop down menu in the software 

### Other information

I will be a good feature to allow more accuracy and less computing power .

nerdCopter, 2024-02-20:
- Some mag require LiPo power.
- Betaflight documentation was recently updated and may be of use. https://betaflight.com/docs/wiki/guides/current/Magnetometer
- Did you cloud-build with Magnetometers?
[image]

spinorose, 2024-02-20:
sure did 
[image]

spinorose, 2024-02-20:
I see information coming out is of the magnetometer  when I connect it using my Arduino to make sure is all working .

haslinghuis, 2024-02-20:
Closing as this is not an issue with configurator. Check status command in CLI.

Duplicate of https://github.com/betaflight/betaflight/issues/13338

closed 2024-02-20 by haslinghuis

spinorose, 2024-02-20:
but my calibration tab is grayed out - so how do I calibrate it then

haslinghuis, 2024-02-20:
Please try our Discord server for support.

spinorose, 2024-02-20:
do you have the discord link

haslinghuis, 2024-02-20:
You can find the [link](https://discord.com/invite/gV4XSq3fus) on the landing page of configurator.