betaflight/betaflight-configurator #3814: Video transmitter page crashes after loading vtx table from json file if FC.VTX_CONFIG.vtx_power value is greater than TABS.vtx.VTXTABLE_POWERLEVEL_LIST

chizar, 2024-02-23:
### Describe the bug

if json file has vtx table that has no current FC.VTX_CONFIG.vtx_power index - UI is crashed due to out of boundary error

### To Reproduce

Video Transmitter page.

1. Add for example 5 power levels.
2. select the latest for Power in "Selected Mode" section (basically set 5 to the FC.VTX_CONFIG.vtx_power)
3. Save
4. Go to Video Transmitter page
5. "Load from file" - upload json file with vtx table, which contains less than 5 power levels
see crashed ui:
[image]

### Expected behavior

no error, 
best scenario is set max FC.VTX_CONFIG.vtx_power

### Configurator version

10.9.0

haslinghuis, 2024-02-23:
1. please provide the json file for testing
2. try configurator 10.10 (nightly build)

chizar, 2024-02-23:
10.10.0 has the issue, as you can see from the attached screenshot (I've made it from the 10.10.0)
json file:
[Readytosky LST 58G VTX.json](https://github.com/betaflight/betaflight-configurator/files/14386675/Readytosky.LST.58G.VTX.json)

nerdCopter, 2024-02-23:
using same .json with Feb 22 softwares.
[image]

chizar, 2024-02-23:
> using same .json with Feb 22 softwares. 

You just missed first step in the step "To Reproduce" section from my original post.
You have to add 5 powers, select the latest in the "Selected Mode", and after that upload the file.

haslinghuis, 2024-02-23:
What does F12 dev console show?

chizar, 2024-02-23:
Sorry, unable to run it via "yarn dev" in browser mode :(
I launch it as "npm start". 
It runs as an application on my mac.

chizar, 2024-02-23:
yep
got it:

[image]

haslinghuis, 2024-02-23:
Will be back at the office in a few days - do not have hardware to test here.

haslinghuis, 2024-02-23:
We need to look at the result in cli after step 3 and after step 5 (with fix applied).

closed 2024-02-23 by haslinghuis