betaflight/betaflight-configurator #3820: New OSD element name

dutkabm, 2024-02-27:
### Is your feature request related to a problem? Please describe

[image]
Currently when add new element to OSD menu it shown as 'Unknow 1'

### Describe the solution you'd like

Would nice to show new unknown elements by default on elements list and on screen like "osd_new_element" -> "New element"

### Describe alternatives you've considered

Na

nerdCopter, 2024-02-27:
You are probably using Configurator 10.9 with a newer Betaflight build.
Please try Configurator 10.10 RC2 or a nightly build.

https://github.com/betaflight/betaflight-configurator/releases/tag/10.10.0-RC2
or
https://github.com/betaflight/betaflight-configurator-nightlies/releases

dutkabm, 2024-02-28:
It's not a issue related versions.  I've added my own new osd element to betaflight firmware. Betaflight configurator see it as "Undefined 1". But I can select it and move on a screen.

nerdCopter, 2024-02-28:
Yes, this is by design. it automatically sees the new elements; however, Configurator code needs to be modified as desired for new elements.
Search pull-requests for `OSD` or `elements` or `OSD elements` for inspiration.

closed 2024-04-06 by github-actions[bot]