betaflight/betaflight-configurator #3821: WebGL: CONTEXT_LOST_WEBGL: loseContext: context lost

haslinghuis, 2024-02-28:
### Describe the bug

[image]

### To Reproduce

Leave configurator open on setup tab (probably also receiver tab)

### Expected behavior

Library not losing context.

### Configurator version

master

### Flight controller configuration

```txt
defaults
```

### Add any other context about the problem that you think might be relevant here

[image]

haslinghuis, 2024-02-28:
[image]

nerdCopter, 2024-02-28:
i found these pertinent links this morning, then quit researching.
seemingly a memory leak, or GPU issue.
https://forum.babylonjs.com/t/what-can-cause-webgl-context-lost/41034 | What can cause webgl context lost? - Questions - Babylon.js
https://github.com/pmndrs/react-three-fiber/discussions/723 | Proper handling of "WebGL: CONTEXT_LOST_WEBGL: loseContext: context lost" and how to restore · pmndrs/react-three-fiber · Discussion #723
https://stackoverflow.com/questions/25219352/webgl-scene-doest-render-because-of-lost-context | three.js - WebGL scene does't render because of lost context - Stack Overflow
https://stackoverflow.com/questions/28135551/webgl-context-lost-and-not-restored#28137949 | javascript - Webgl Context Lost and not restored - Stack Overflow

PR #3824 (2024-03-01): Fix lost context
Changes

- Fix calling render functions while disposed
- Do not double precision in `renderer.setSize`
- Fixes #3821 (at least to seem to occur less frequently)
- See: https://stackoverflow.com/questions/25219352/webgl-scene-doest-render-because-of-lost-context