betaflight/betaflight-configurator #3830: Osd not working when using a vista and goggles v2 with wftos installed

viseofpv, 2024-03-08:
### Describe the bug

Missing gps sats, link quality and others in the goggles v2 with wtfos installed and using caddx vista but every works when going back to 4.4.3 ? 

### To Reproduce

turn on goggles and air unit vista

### Expected behavior

everything working in the OSD

### Configurator version

10.10.10

RoadyFPV, 2024-03-09:
maybe this could help

set displayport_msp_fonts = 0,0,0,0

Canaill51, 2024-03-09:
> ### Describe the bug
>
> Missing gps sats, link quality and others in the goggles v2 with wtfos installed and using caddx vista but every works when going back to 4.4.3 ?
> ### To Reproduce
>
> turn on [...]

Did you make sure to apply correct HD VTX preset?
[image]