betaflight/betaflight-configurator #3833: Flash local firmware not working since 10.9

Berlinologe86, 2024-03-11:
### Describe the bug

I had a problem copter on the table on Saturday and wanted to install the firmware from the manufacturer. Betafpv Pavo pico/pavo 20. (bf4.4.1) I couldn't do this on either Windows or Mac. Cloud building worked on both systems. The configurator never switched to the bootloader and flashed local hex files. Not even with the bootloader activated. The last attempt was to try out 10.8.0. It worked wonderfully with that. Kind regards, Sören

### To Reproduce

Try local firmware upload with 10.9 and 10.8

### Expected behavior

A Little fix ?

### Configurator version

10.8/10.9/10.10

nerdCopter, 2024-03-11:
Although this document mainly for Windows, please investigate all notes: https://hackmd.io/@nerdCopter/rJv5TUrQ2

Third party dependencies may be involved here (NWjs upon which the UI is based.)

Berlinologe86, 2024-03-11:
I think you didn't understand me correctly. On Windows and Mac, the local firmware upload only works without problems from version 10.8. This doesn't work with the newer versions 10.9 and 10.10. A system error can therefore be ruled out. Since it works as usual with 10.8. So I don't have a driver problem. The configurator has a problem.

josser, 2024-03-20:
Same for me for Meteor65 Pro but not working even on 10.8 
1. Opened Configurator. (10.8, 10.9, 10.10-RC3) (Mac, Windows 11)
2. Selected target BETAFPVF411
3. Selected local file. Configurator saying: "Loaded Local Firmware"
4. Clicked "Flash Firmware" and nothing happened. 
[image]
5. Online flashing works fine, but I want to flash OEM firmware because it has some extra fixes related to OSD. With original Betaflight meteor65 pro has some blinking on 4.3+

nerdCopter, 2024-03-21:
There may be a bug in some unknown state of workflow.
I suggest running Configurator fresh, and immediately Load-Local without doing anything else first. do not even select any items in drrop-down fields.
[image]
[image]

haslinghuis, 2024-03-21:
Have tested local flashing succeeded.

josser, 2024-03-21:
Could it be somehow related to the "custom" nature of the .hex file? 
It was built by BetaFPV and I bet they made some changes to it.

nerdCopter, 2024-03-21:
Tested PYRODRONEF7 with 10.9 and two different local .hex -- erased/flashed successfully on Linux.
Tested same with 10.10.-RC3, successfully on Linux. 

**_May be a timing issue, or something specific to local computers._** :man_shrugging: :confused: 

<hr>

Further notes:
10.10-RC3 auto-reconnected after the v4.5 reboot as well; However there was no reboot with the v4.4.3.

4.5 + 10.10-RC3:
```
2024-03-21 @08:28:42 -- Erased 512 kB of flash successfully
2024-03-21 @08:29:03 -- USB device successfully closed
2024-03-21 @08:29:05 -- Query board information to preselect right firmware
2024-03-21 @08:29:05 -- MultiWii API version: 1.46.0
2024-03-21 @08:29:05 -- Configurator has successfully detected and verified the board: PYRODRONEF7
```
auto-reconnected and shows limited buttons:
[image]

nerdCopter, 2024-03-21:
> Could it be somehow related to the "custom" nature of the .hex file? It was built by BetaFPV and I bet they made some changes to it.

Interesting!Yes, if HEX fails parsing/checks, it may not enable Flashing.

josser, 2024-03-21:
https://github.com/betaflight/betaflight-configurator/assets/103639/8f1001ee-1f68-4960-bfab-758927b42d93

Video demo

nerdCopter, 2024-03-21:
please post a link to the BetaFPV hex.

josser, 2024-03-21:
https://support.betafpv.com/hc/en-us/article_attachments/13093661336729

nerdCopter, 2024-03-21:
Hex problem, IMHO. 

[image]

josser, 2024-03-21:
Hm, that's correct, BetaFPV provides CLI-dump but tells 
"CLI for quadcopter (Do not need to be flashed separately):"
Is there any way to bypass this check?

Berlinologe86, 2024-03-21:
https://support.betafpv.com/hc/en-us/articles/21133140518937-CLI-and-Firmware-for-Pavo-Pico-F405-2-3S-20A-V1-0-FC

nerdCopter, 2024-03-21:
i don't know for fact, but i think that the hex points to an incorrect memory address. nothing to do with the .txt file even though i did experiment with it on the second try.

Berlinologe86, 2024-03-21:
https://support.betafpv.com/hc/en-us/articles/23244448530201-CLI-and-Firmware-for-Pavo-20-F405-2-3S-20A-V1-0-FC

nerdCopter, 2024-03-21:
> [support.betafpv.com/hc/en-us/articles/23244448530201-CLI-and-Firmware-for-Pavo-20-F405-2-3S-20A-V1-0-FC](https://support.betafpv.com/hc/en-us/articles/23244448530201-CLI-and-Firmware-for-Pavo-20-F405-2-3S-20A-V1-0-FC)

same result. thank you both. someone else with further knowledge will need to take-over. or maybe BetaFPV need to be informed.?
[image]

josser, 2024-03-22:
I've contacted BetaFPV support and they told me the following: 

>  We need to use BF 10.8.0 to flash the custom firmware, and if we encounter a stuck situation, we can solve it by restarting BF, as shown in the video.
> https://drive.google.com/file/d/1iirpJmA30ne1AWNkYhNOM5qIPuDGoaXC/view?usp=drive_link

I've tried this and everything is worked! 

I think it's solution for all betafpv bnf quads.

Now I'm not sure is it a bug or it's expected behaviour ) 
But this issue is helpful for those who will face same issues.

haslinghuis, 2024-03-22:
[image]

haslinghuis, 2024-03-22:
For reference: this was the `diff` the board came with. ACC + GYRO were not recognized. After flashing 4.5RC3 it works.

[BTFL_cli_PAVO_PICO_20240322_171322_BETAFPVF405.zip](https://github.com/betaflight/betaflight-configurator/files/14725602/BTFL_cli_PAVO_PICO_20240322_171322_BETAFPVF405.zip)

closed 2024-03-22 by haslinghuis