betaflight/betaflight-configurator #3838: AT32F435 do not response after some changes on MACOS

atomadog, 2024-03-18:
### Describe the bug

AT32F435 (neutronrc435se) do not response after some changes on MACOS, need reboot, but ok on Windows

### To Reproduce

when disable/enable any rx, chose input sensor, change arming angle and etc, dont response after click save/reboot button, need power off and reconnect. I tryid it on different macbooks. on Windows no problem

### Expected behavior

fix it on macos pls

### Configurator version

10.10.0-debug-0e061df

### Flight controller configuration

```txt
# 

# diff all

# version
# Betaflight / AT32F435G (A435) 4.5.0 Dec 25 2023 / 10:48:35 (22ce58655) MSP API: 1.46
# config rev: 9bf017a

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name NEUTRONRCF435SE
manufacturer_id NERC
mcu_id 1049c13804f5c0000d6d9717
signature 

# feature
feature -LED_STRIP
feature GPS
feature TELEMETRY
feature ESC_SENSOR

# serial
serial 0 131073 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 3 2 115200 57600 0 115200
serial 4 1024 115200 57600 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# led
led 0 6,12::CTYBV:0
led 1 7,12::CTYBV:0
led 2 8,12::CTYBV:0
led 3 9,12::CTYBV:0

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 2 1275 1675 0 0
aux 2 46 2 1650 2100 0 0
aux 3 13 5 1575 2100 0 0
aux 4 35 4 1650 2100 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 BAND_D   D CUSTOM  5362 5399 5436 5473 5510 5547 5584 5621
vtxtable powerlevels 3
vtxtable powervalues 25 100 200
vtxtable powerlabels 100 800 120

# master
set gyro_hardware_lpf = OPTION_2
set gyro_lpf1_static_hz = 0
set dyn_notch_q = 500
set gyro_lpf1_dyn_min_hz = 0
set gyro_lpf1_dyn_max_hz = 0
set acc_calibration = -15,62,-8,1
set rc_smoothing_auto_factor = 140
set rc_smoothing_setpoint_cutoff = 25
set rc_smoothing_feedforward_cutoff = 25
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set failsafe_delay = 10
set failsafe_procedure = GPS-RESCUE
set ibata_scale = 1000
set yaw_motors_reversed = ON
set small_angle = 180
set gps_rescue_throttle_hover = 1300
set gps_rescue_min_sats = 6
set gps_rescue_allow_arming_without_fix = ON
set simplified_gyro_filter = OFF
set report_cell_voltage = ON
set osd_vbat_pos = 2437
set osd_rssi_pos = 2077
set osd_tim_1_pos = 3466
set osd_tim_2_pos = 3498
set osd_flymode_pos = 2361
set osd_throttle_pos = 3530
set osd_vtx_channel_pos = 454
set osd_current_pos = 2468
set osd_mah_drawn_pos = 2502
set osd_gps_speed_pos = 2552
set osd_gps_lon_pos = 3073
set osd_gps_lat_pos = 2056
set osd_gps_sats_pos = 2069
set osd_home_dir_pos = 2490
set osd_home_dist_pos = 2520
set osd_altitude_pos = 3434
set osd_warnings_pos = 2581
set osd_avg_cell_voltage_pos = 2392
set osd_nvario_pos = 3402
set osd_esc_tmp_pos = 2405
set osd_displayport_device = MSP
set debug_mode = GYRO_SCALED
set vcd_video_system = HD

profile 0

# profile 0
set vbat_sag_compensation = 100
set p_pitch = 51
set i_pitch = 92
set d_pitch = 55
set f_pitch = 137
set p_roll = 49
set i_roll = 88
set d_roll = 48
set f_roll = 132
set p_yaw = 49
set i_yaw = 88
set f_yaw = 132
set d_min_roll = 36
set d_min_pitch = 41
set thrust_linear = 20
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 45
set feedforward_jitter_factor = 10
set dyn_idle_min_rpm = 30
set simplified_master_multiplier = 110
set simplified_d_gain = 110

profile 1

profile 2

profile 3

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#
```

haslinghuis, 2024-03-18:
Duplicate of #3514

Yamboss, 2024-05-16:
any news for mac at32 users ? did anyone resolve this some other way ?