betaflight/betaflight-configurator #3839: BF GUI RC3 - Presets not loading

Canaill51, 2024-03-19:
### Describe the bug

From BF GUI RC3 on Android app, Presets are not being loaded

See the video as attached.
https://youtu.be/3gEgdT-oNEc

### To Reproduce

Install BF GUI RC3 on an Android Smartphone
Connect to FC flashed on BF 4.5
Go Presets Tab and select one (ex: HD, ELRS, etc...).

### Expected behavior

Presets selected should be downloaded to apply then to FC.

### Configurator version

betaflight-configurator_10.10.0-RC2-debug-norevision

haslinghuis, 2024-03-19:
Connected with FC the preset tiles show up here.

Going back to nightly builds to find the culprit - now on 29 january - still gives the `Error loading presets from the internet` error message.

ASDosjani, 2024-03-19:
I tested the nightly android build. On windows it's fine.

haslinghuis, 2024-03-19:
Debugging android shows the following the download has been blocked by CORS policy:

[image]

ASDosjani, 2024-03-19:
We fixed this issue a while back😂

ASDosjani, 2024-03-19:
I added cleartext support in this PR:
https://github.com/betaflight/betaflight-configurator/pull/3095/files#diff-2431b1e97cd2951b5d8333ad7094652ff5f7131b63509116e2531122a2d8a94a

Canaill51, 2024-03-19:
Do you guys, need anything else from me?

limonspb, 2024-03-19:
> Do you guys, need anything else from me?

Could you please test if backup preset source works? (Deactivate the main, and activate backup)

haslinghuis, 2024-03-19:
Maybe @atomgomba might know a solution

ASDosjani, 2024-03-19:
Backup works for me

Canaill51, 2024-03-20:
> > Do you guys, need anything else from me?
> 
> Could you please test if backup preset source works? (Deactivate the main, and activate backup)

Yes, it's working. Just tried / tested:
https://youtu.be/K0TkzgZqJzo?si=8_VeiS3FeUup0qEn

However, "Backup" is still available up to 4.4 firmware.

Canaill51, 2024-03-21:
@HThuren , not sure why you assigned this issue to me :)

Canaill51, 2024-04-06:
Hi guys,
What are the next steps for this? Will it be fixed when Released?

haslinghuis, 2024-04-06:
@Canaill51 Seems to work now for me now. There was an issue with CORS policy...

SupaflyFPV, 2024-04-06:
Tested using this config on old android phone

https://github.com/betaflight/betaflight-configurator/releases

Works as expected for me.

@Canaill51 can you confirm?

closed 2024-04-06 by haslinghuis

Canaill51, 2024-04-06:
@haslinghuis, @SupaflyFPV , just tried and tested: Works now!

Thanks all for your help (as always!). Really appreciate