betaflight/betaflight-configurator #3843: Can't connect to FC with MacOS Sonoma 14.3 and MAMBA Stack F722 MK3

Jannis12324, 2024-03-19:
### Describe the bug

Hello :) 
I can't connect to my Flight Controller a MAMBA Stack F722 MK3 through Beta Flight.
When I plug in the USB C cable nothing happens in betaflight.

### To Reproduce

Setup I'm using:
- MAMBA Stack F722 MK3 Flight Controller
- Mac Book Air M2 with Mac OS Sonoma 14.3
-[ Data USB cable](https://www.amazon.de/dp/B07VGMCT1J?psc=1)

Steps I took:
- Installed betaflight and allowed it to run in the Mac Privacy and security settings
- Opened Betaflight
- Installed the STM USB VCP Drivers for Mac OS linked in Betaflight
- connected the FC
- (nothing happens)
- Restarted the Laptop multiple times, 
- checked security settings if it needs to be allowed again and 
- reinstalled drivers

Nothing works.

Assumptions why it might not work:
- The USB-C port doesnt provide enough power: As it can charge a phone I don't think that's an issue
- It doesn't work with the new Mac OS Sonoma: So far I couldn't find any hints on that
- The drivers are so old (2019 last updated) that they don't work with Sonoma
- Betaflight of the driver is blocked somewhere by the OS: Some youtube video mentioned it has to be unblocked multiple times, but I was only prompted to unblock once.

Thanks for any help !

### Expected behavior

The FC should show up, once it's connected.

### Configurator version

10.9.0

nerdCopter, 2024-03-19:
- Try https://github.com/betaflight/betaflight-configurator/releases/tag/10.10.0-RC3 .
- USB-C may need an adapter or OTG cable.
- Try https://app.betaflight.com/ in Google Chrome (but no flashing in Web App yet).

haslinghuis, 2024-03-19:
There is no need to install the STM USB VCP Drivers on Mac OS or any other drivers.

Jannis12324, 2024-03-24:
@nerdCopter I tried the release https://github.com/betaflight/betaflight-configurator/releases/tag/10.10.0-RC3
and https://app.betaflight.com/ but none find my FC. It says "no compatible devices can be found"

> USB-C may need an adapter or OTG cable.
Why is that the case? The FC and my Laptop both have USB-C.

Stupid question: Do I have to connect the FC to the battery? I assumed it would get enough power from the USB C cable to run

Jannis12324, 2024-03-24:
> There is no need to install the STM USB VCP Drivers on Mac OS or any other drivers.

Ah didn't know, thanks. I uninstalled it.

nerdCopter, 2024-03-25:
> Why is that the case? The FC and my Laptop both have USB-C.

I don't know why, but many pilots found this to be the fact. My Dell/Linux USB-C works fine when i plug FC into a hub, but if i go direct to PC's USB-C, then i need a dongle adapter.  Unsure why, but my USB-C to USB-C wires never work. Maybe other pilots have different experiences. :man_shrugging: 

> Stupid question: Do I have to connect the FC to the battery? I assumed it would get enough power from the USB C cable to run

Do not need for battery.

closed 2024-05-02 by github-actions[bot]