betaflight/betaflight-configurator #3844: Change TPA rate and Breakpoint to Percentages for ease of understanding

ChrisRosser, 2024-03-20:
### Is your feature request related to a problem? Please describe

For newer pilots TPA rate and breakpoint are somewhat obscure. It's not always clear to them that throttle goes from 1000 to 2000 as this isn't really the case any more with Dshot. 1350 would become 35%.

TPA rate is clearer but might make even more sense as a percentage attenuation. 0.65 would become 65%.

### Describe the solution you'd like

Presenting TPA rate and TPA breakpoint as percentages (and perhaps converting to 1000-2000 behind the scenes) and adding a tooltip explaining TPA would resolve this problem.

### Describe alternatives you've considered

I don't see the need to change anything in BF I think this could just be a configurator tweak.

nerdCopter, 2024-03-21:
@ChrisRosser , my opinion is only a personal one. If the PR i made is not acceptable, then you can certainly open debate for other devs' attention. thank you.

closed 2024-03-21 by haslinghuis

haslinghuis, 2024-03-21:
Is it okay now @ChrisRosser  ?

ChrisRosser, 2024-03-22:
Hi Mark,

TPA rate is now very clear thank you!

I still take issue with using us for throttle.

1. Motor output limit on the same page is a percentage.

2. No-one uses PWM for throttle anymore.

Thanks,

Chris

haslinghuis, 2024-03-22:
@ChrisRosser did a take 3 - see above - but it does not correlate with blackbox. See comment in PR.

PR #3846 (2024-03-21): TPA clarification - % and μs
- resolves #3844
- will need Crowdin translations

[image]