betaflight/betaflight-configurator #3849: missing or iinvlaid `initialSetupInfoBuildOptionsEmpty`

nerdCopter, 2024-03-21:
### Describe the bug

[image]

```shell
## master...origin/master
$ grep -Iirn initialSetupInfoBuild.*Empty ./locales/
./locales/ko/messages.json:1039:    "initialSetupInfoBuildEmpty": {
./locales/uk/messages.json:1035:    "initialSetupInfoBuildEmpty": {
./locales/da/messages.json:1039:    "initialSetupInfoBuildEmpty": {
./locales/pl/messages.json:1039:    "initialSetupInfoBuildEmpty": {
./locales/pt_BR/messages.json:1035:    "initialSetupInfoBuildEmpty": {
./locales/ca/messages.json:1043:    "initialSetupInfoBuildEmpty": {
./locales/en/messages.json:1119:    "initialSetupInfoBuildEmpty": {
./locales/es/messages.json:1035:    "initialSetupInfoBuildEmpty": {
./locales/gl/messages.json:1038:    "initialSetupInfoBuildEmpty": {
./locales/ja/messages.json:1035:    "initialSetupInfoBuildEmpty": {
./locales/pt/messages.json:1035:    "initialSetupInfoBuildEmpty": {
./locales/it/messages.json:1035:    "initialSetupInfoBuildEmpty": {

## master...origin/master
$ grep -Iirn initialSetupInfoBuild.*Empty ./src/
./src/js/tabs/setup.js:450:                    build_info_e.html(i18n.getMessage(navigator.onLine ? 'initialSetupInfoBuildOptionsEmpty' : 'initialSetupNotOnline'));
./src/js/tabs/setup.js:470:                    build_info_e.html(i18n.getMessage('initialSetupInfoBuildEmpty'));

```

### To Reproduce

master

### Expected behavior

lang

### Configurator version

master

nerdCopter, 2024-03-21:
@HThuren, want this task?

haslinghuis, 2024-03-21:
It's in `locales/en/messages.json` so it up to crowdin to fix.

nerdCopter, 2024-03-21:
> It's in `locales/en/messages.json` so it up to crowdin to fix.

no, the javascript has `initialSetupInfoBuildOptionsEmpty` , but the `en` only has `initialSetupInfoBuildEmpty`.  i think `Options` needs to be removed, but not 100% certain.

PR #3850 (2024-03-21): Fix missing message
- Fixes #3849
closed 2024-03-21 by haslinghuis