betaflight/betaflight-configurator #3856: In column of Power & Battery, there is no fields to fill Voltage & Current when i click the "calibrate" button.

Signal99, 2024-03-23:
### Describe the bug

In column of Power & Battery, there is no fields to fill Voltage & Current when i click the "calibrate" button. 
[image]

### To Reproduce

Every time i click the button of "Calibrate", it happens.

### Expected behavior

Calibrate Voltage & Current.

### Configurator version

10.9

haslinghuis, 2024-03-23:
[image]

After applying power to the ESC and rebooting FC:
[image]

Signal99, 2024-03-24:
Thanks for ur reply. finally, i got the answer. 
actually, i powered the ESC. But i didnt calibrated the Voltage manually, so the ESC got a wrong voltage, it seems like hav no battery.  
thanks.

closed 2024-03-24 by Signal99