betaflight/betaflight-configurator #3860: No response after launching betaflight configurator

ClementYap, 2024-03-26:
### Describe the bug

No response when Betaflight configuration is launched.

### To Reproduce

In Ubuntu software, I downloaded  and installedBetaflight Configurator. when searching BetaFlight in Ubuntu Software, the status showed that it is installed. when this not worked, I followed instruction to download - "Clone Betaflight Repository and Install Toolchain"

### Expected behavior

It would be opened as expected in my window 10.

### Configurator version

Can't read

ClementYap, 2024-03-27:
> Is your Configurator directly from
> https://github.com/betaflight/betaflight-configurator/releases ?
> For Ubuntu, use the .deb files.
> From a commandline, uninstall the old version with this command: sudo apt
> purge betaflight-configurator
> Then once downloaded, install the new version with the following command sudo
> dpkg -i betaflight-configurator_10.10.0-RC3_amd64.deb from the folder in
> which it is downloaded.
>
> if it still does not execute, list the file attributes with the command
> and report results: ls -lh
> /opt/betaflight/betaflight-configurator/betaflight-configurator*

Hi. NerdCopter,
Thanks

I am unable to find betaflight-configurator_10.10.0-RC3_amd64.deb in github.
Is it because it is pre-releas?

I got these non-working beta-connfiguratora from Ubuntu Software and apt
command.

regards
 Clement.

nerdCopter, 2024-03-27:
[image]

ClementYap, 2024-03-28:
Hi, nerdCopter. Thanks. Found it.
I am able to install the downloaded betaflight-configurator.

However It is stopped by the requirement for the installation of
gconf2-common_3.2.6-5_all.deb and libgconf-2-4.3.2.6-5_amd64.deb too.

I am not able to download them as I do not have a business email.

Please advise. Thanks
Clement Yap

nerdCopter, 2024-03-28:
maybe you can install with `apt` to auto-install dependencies, i'm not 100% certain. 
`sudo apt install -f ./betaflight-configurator_10.10.0-RC3_amd64.deb`

if that doesn't work well enough, check this: https://askubuntu.com/a/40050/570833
i.e.
```
sudo dpkg -i betaflight-configurator_10.10.0-RC3_amd64.deb
sudo apt install -f
```

ClementYap, 2024-03-30:
Hi, nerdCopter.
 Thanks. Your help is working.

regards
 Clement

haslinghuis, 2024-03-30:
@ClementYap want to add this to the README. Which commands did work out for you?

```
sudo apt install -f ./betaflight-configurator_10.10.0-RC3_amd64.deb
```

or 

```
sudo dpkg -i betaflight-configurator_10.10.0-RC3_amd64.deb
sudo apt install -f
```

haslinghuis, 2024-03-30:
See: https://github.com/betaflight/betaflight-configurator/?tab=readme-ov-file#linux-users

ClementYap, 2024-03-31:
Hello, @haslinghuis 
The first one is working. After downloading the program as shown in the instruction above, Run the first sudo command from the folder where it is downloaded.

ClementYap, 2024-03-31:
> Wonder if this would have resolved the issue:
> 
> ```
> sudo apt install libatomic1
> ```
> 
> See: https://github.com/betaflight/betaflight-configurator/?tab=readme-ov-file#linux-users

That one I have not tried. Thanks. Now it is working and running smmothly.

closed 2024-05-05 by haslinghuis