betaflight/betaflight-configurator #3866: Failsafe modification

alex39292, 2024-03-30:
### Is your feature request related to a problem? Please describe

When using the "landing" in Failsafe configuration page, I have to specify the time for the drone to descend.  But the drone may not have enough time and will simply turn off the motors without landing.

### Describe the solution you'd like

On flight controllers with a barometer installed, we can take changes in altitude data to determine whether the drone has landed or not.  I propose to add a radio button to the “Landing” tab: “Landing by barometer” and “Delay for turning off motors after landing” [10] seconds - i.e.  Altitude data does not change for [10 ]seconds.
[image]

### Describe alternatives you've considered

We need to consider the best way for the real drone landing during the failsafe procedure.

haslinghuis, 2024-03-30:
See also https://github.com/betaflight/betaflight/issues/4663 :)

alex39292, 2024-04-17:
Yeah. I didn't see it ) The goal is to determine changes in baro data for example each 1 second ( changes in vertical speed). If no changes - our quad landed.

haslinghuis, 2024-05-05:
Duplicate of #4663

closed 2024-05-05 by haslinghuis