betaflight/betaflight-configurator #3868: Reciever Tab model preview not responding.

El-Takanashi, 2024-04-01:
### Describe the bug

Even though the channel bars respond correctly to stick/switch inputs, the preview of the model does not move at all. Hakrc f722v2 and SpeedyBee f405v3. reverting to 10.9 shows model preview move correctly.

### To Reproduce

Power radio. Connect to config 10.10.0 r3. Go to receiver tab,  move sticks - observe bars move correctly, but not model preview at all.

### Expected behavior

model preview to move in accordance with stick inputs/channel bars

### Configurator version

10.10.0

haslinghuis, 2024-04-01:
Duplicate of #3853

haslinghuis, 2024-04-29:
- Please use configurator [release](https://github.com/betaflight/betaflight-configurator/releases/tag/10.10.0)
- Should be fixed in #3847

closed 2024-05-03 by haslinghuis