betaflight/betaflight-configurator #3869: Failed to install 10.10.0-RC3 .deb on Ubuntu 23.10 because of dependencies

amatsumara, 2024-04-02:
### Describe the bug

I downloaded latest release and was trying to install it:
```
dpkg -i betaflight-configurator_10.10.0-RC3_amd64.deb 
Selecting previously unselected package betaflight-configurator.
(Reading database ... 352261 files and directories currently installed.)
Preparing to unpack betaflight-configurator_10.10.0-RC3_amd64.deb ...
Unpacking betaflight-configurator (10.10.0-RC3) ...
dpkg: dependency problems prevent configuration of betaflight-configurator:
 betaflight-configurator depends on libgconf-2-4; however:
  Package libgconf-2-4 is not installed.

dpkg: error processing package betaflight-configurator (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 betaflight-configurator
```

### To Reproduce

Download [10.10.0-RC3](https://github.com/betaflight/betaflight-configurator/releases/tag/10.10.0-RC3) .deb package from releases section. Run:
`dpkg -i betaflight-configurator_10.10.0-RC3_amd64.deb`

### Expected behavior

At least .deb linked to existing Ubuntu package so it is possible to install it but there none:
```
apt search libgconf-2-4
Sorting... Done
Full Text Search... Done
```

### Configurator version

10.10.0-RC3

haslinghuis, 2024-04-02:
Please check [README](https://github.com/betaflight/betaflight-configurator?tab=readme-ov-file#linux-users)

haslinghuis, 2024-04-02:
Specific

> The libatomic library must also be installed before installing Betaflight Configurator. (If the library is missing, the installation will succeed but Betaflight Configurator will not start.) Some Linux distributions (e.g. Fedora) will install it automatically. On Debian or Ubuntu you can install it as follows:
```
sudo apt install libatomic1

```

McGiverGim, 2024-04-03:
I'm more a windows user, so I don't know if I understand exactly the issue, but the dependencies, must be installed automatically:

https://github.com/betaflight/betaflight-configurator/blob/eb8f3986c54e697bda1dc795db177312308ca940/gulpfile.js#L802

the problem is that dpkg doesn't do that. You must install the dependencies by hand. Nothing we can do with it as far as I know.

nerdCopter, 2024-04-03:
- this is basically a duplicate of: https://github.com/betaflight/betaflight-configurator/issues/3860
- solutions there as well.

haslinghuis, 2024-04-03:
See: https://2h3ph3rd.medium.com/how-to-install-libgconf-2-4-on-ubuntu-23-10-fec6bda8d5f5

amatsumara, 2024-04-03:
> * this is basically a duplicate of: [No response after launching betaflight configurator #3860](https://github.com/betaflight/betaflight-configurator/issues/3860)
> 
>     * solutions there as well.

There is no information of Ubuntu version used in that issue. **libatomic1** installed already in Ubuntu 23.10. 
**apt install -f** last lines output:
```
The following packages have unmet dependencies:
 betaflight-configurator : Depends: libgconf-2-4 but it is not installable
E: Unable to correct problems, you have held broken packages.
```
**libgconf-2-4** can't be found in Ubuntu 23.10 repos.

nerdCopter, 2024-04-03:
Thanks for clarification @amatsumara , apologies for my not understanding Ubuntu 23.x has updated libraries.

nerdCopter, 2024-04-03:
i was able to remove `libgconf-2-4` as a dependency in the gulpfile.js, and install into a fresh Xubuntu 23.10 VM.
i did not test removing `libatomic1`.
further testing should be done on legacy systems i suppose.

haslinghuis, 2024-04-03:
libgconf-2-4: This package is for legacy applications and no longer used by GNOME. [Link](https://packages.debian.org/buster/libgconf-2-4)

Agree we might to keep this in for backwards compatibility.

haslinghuis, 2024-04-04:
Afaik Betaflight does only support LTS versions of [Ubuntu](https://wiki.ubuntu.com/Releases)

The missing package could be installed using https://2h3ph3rd.medium.com/how-to-install-libgconf-2-4-on-ubuntu-23-10-fec6bda8d5f5 as guess we need to keep it for Ubuntu 20.04 LTS for at least one year.

nerdCopter, 2024-04-04:
Good choice. I also use only LTS. If it can be gated in some way, such would be the way to go.

amatsumara, 2024-04-04:
> Afaik Betaflight does only support LTS versions of [Ubuntu](https://wiki.ubuntu.com/Releases)
> 
> The missing package could be installed using https://2h3ph3rd.medium.com/how-to-install-libgconf-2-4-on-ubuntu-23-10-fec6bda8d5f5 as guess we need to keep it for Ubuntu 20.04 LTS for at least one year.

Do you think all will work well in Ubuntu 24.04 LTS Noble Numbat (should be released in April 2024)? Will try it.

Yeah, installing Debian repos should work.

haslinghuis, 2024-04-04:
We will check next LTS version after release.

vitali2y, 2024-04-07:
Yeah, [this recept](https://gist.github.com/vitali2y/950bc3ad341b5a602c0a97cac346ca80) works as a charm @ _Linux Mint 21.2_.

PR #3884 (2024-04-10): Add additional installation steps for Ubuntu 23.10
- Fixes #3869
closed 2024-04-09 by amatsumara