betaflight/betaflight-configurator #3874: OSD Broken in 4.5.0+

ericgever, 2024-04-05:
### Describe the bug

Attempting to load a development firmware onto a MatekF405, noticed the OSD no longer works in any 4.5.0 dev build (including release candidates and master).  Unable to show splash screen or any parameters.  When I revert to 4.4.3 and diff my configuration to it, OSD works immediately.  Also, Font Manager is grayed out on 4.5.0, but able to configure font on 4.4.3 and earlier.

### To Reproduce

Attempt to view OSD (on analog VTX/camera) using 4.5.0 dev build

### Expected behavior

OSD should show...

### Configurator version

10.9.0

haslinghuis, 2024-04-05:
Use latest configurator to support firmware 4.5 - and between versions a previous diff is not always compatible.

closed 2024-04-05 by haslinghuis

nerdCopter, 2024-04-05:
> [...] Font Manager is grayed out on 4.5.0 [...] 

i've seen this as well, please check if it is available when LiPo plugged.
if it is not working, then the code is not detecting max7456 properly. (`requires-max7456-font-device-detected` in the code.)

flux242, 2024-04-08:
what's the equivalent cli command line? That would not work
```
make MY_FC_HW_CONF EXTRA_FLAGS="-DUSE_OSD_SD"
```

nerdCopter, 2024-04-08:
> what's the equivalent cli command line? 

Use Configurator to build, then click the "Log" link. The log will provide full commandline.  remove the build-key from that command when building local.

flux242, 2024-04-08:
> > what's the equivalent cli command line?
> 
> Use Configurator to build, then click the "Log" link. The log will provide full commandline. remove the build-key from that command when building local.

log link doesn't work for me. Will you put an example here?

nerdCopter, 2024-04-08:
> log link doesn't work for me. Will you put an example here?

Discord for further support please.
[image]
https://build.betaflight.com/api/builds/e688d4d897c94f027824b533ba4a767e/log
[image]

Pitfirst, 2024-05-01:
Hi, I has a similar issue. After selecting the font "Betaflight" instead if "standard", everything went perfect.

bd4, 2024-05-02:
I am seeing this issue as well, with configuration 10.10, betaflight 4.5 with OSD (SD) only, on a Cetus X with analog vtx. I tried changing font with no change. Font Manager is greyed out.

Rolling back to 4.4.3 restores the OSD and font manager button.

nerdCopter, 2024-05-02:
re: `Font Manager is greyed out`.  seems the code has max7456 dependency that is not being respected. 
`requires-max7456-font-device-detected`
[image]

bladan83, 2024-05-11:
Mamb f722 same problem, OSD is not visible and font manager is grayed out

nerdCopter, 2024-05-13:
> Mamba f722 same problem after update 4.5, OSD is not visible and font manager is grayed out

1) for "OSD not visible", be certain to exclude the `OSD (HD)` option from build options and only use `OSD (SD)` for analog.
2) i finally circumvented my `Font manager` grayed out.  i had to toggle full-chip erase a couple times, and i also manually went into DFU mode via boot-button. apparently some old settings were lingering. 
 - reference: https://github.com/betaflight/betaflight-configurator/issues/3929

bladan83, 2024-05-13:
for OSD not visible I solved useding this command "set displayport_msp_fonts = 0,0,0,0", it seems to fix the problem, now the OSD is showed as well. I make a difference of diff file before and after the fix, i found other 2 new commands on the fix version: "set osd_warn_bitmask = 327679" and "set osd_displayport_device = AUTO"

nerdCopter, 2024-05-13:
> For gray button, right now I don't need to change font, in the nex update i will try

`Font manager` is only for onboard MAX7456 analog OSD chip.

Neil-Sawhney, 2024-06-02:
Same issue where OSD no longer works on 4.5.0. Only `OSD (SD)` is selected on flash. The font manager is not greyed out but changing fonts does not solve the issue.

Neil-Sawhney, 2024-06-03:
> Same issue where OSD no longer works on 4.5.0. Only `OSD (SD)` is selected on flash. The font manager is not greyed out but changing fonts does not solve the issue.

solved the issue. Seemed to be caused by my backup parameters. Fixed it by doing a fresh flash and doing a diff against my backup. Removed all the changed osd related stuff besides element positions and it started working again.

Lihatskiy, 2024-06-07:
For analog video on OSD settings page set video format NTSC or PAL.
It solved issue.

haslinghuis, 2024-06-07:
@ericgever also make sure to be using configurator 10.10

pitev, 2024-07-10:
> for OSD not visible I solved useding this command "set displayport_msp_fonts = 0,0,0,0", it seems to fix the problem, now the OSD is showed as well. I make a difference of diff file before and after the fix, i found other 2 new commands on the fix version: "set osd_warn_bitmask = 327679" and "set osd_displayport_device = AUTO"
> 
> For gray button, right now I don't need to change font, in the nex update i will try

worked for me as well, but only w/  "set osd_warn_bitmask = 327679" and "set osd_displayport_device = AUTO" added to cli

dmak, 2025-01-12:
In my case I had to `set osd_displayport_device = AUTO` after upgrading SpeedyBee F405 V4 to Betaflight 4.5.1 to make OSD working. Thanks for suggesting it!

jeaton2411, 2025-02-04:
I know this is closed but the issue persists with me. I tried "set displayport_msp_fonts = 0,0,0,0", as well as "set osd_warn_bitmask = 327679" and "set osd_displayport_device = AUTO"

First on the 4.5 build that came on my TBS Lucid AIO then after a clean rebuild of 4.5.1 on the TBS board with OSD (HD) only build option. Have also tryed the above fixes on 4.5.1

I have triple checked the wiring.