betaflight/betaflight-configurator #3880: 3D drone under "setup" sometimes does not show motion on the yaw axis.

SentinelFour, 2024-04-07:
### Describe the bug

Occasionally, the 3D image of the drone under "Setup" does not show any movement on the yaw axis. Pitch and roll work fine. To fix this, either recalibrate the accelerometer or replug/power cycle the drone.
This happens rarely, but with all drones (from whoops to 5 inch and with different FCs). I also had this problem with the previous version of the configurator. Currently on the latest release candidate and it still happens. Just wondering what is causing this or if there is anything I can do on my end.

### To Reproduce

Hard to reproduce consistently. I would say it happens 1 out of 10 times when connecting a drone. I haven't found a procedure that reliably prevents this. 

### Expected behavior

Well, all three axes show correct movement - all the time :)

### Configurator version

10.10.0-RC3 (but happened also with 10.9)

### Add any other context about the problem that you think might be relevant here

I checked Discord but found no information about this. Google didn't turn up much either, but there have been a few reports:
https://www.reddit.com/r/Multicopter/comments/9fe6ny/quad_image_doesnt_yaw_in_betaflight_configurator/
https://www.reddit.com/r/fpv/comments/vb0r1i/help_yaw_is_broken_in_betaflight/

haslinghuis, 2024-04-07:
Check the `CALIB` flag while it happens in setup tab.

SentinelFour, 2024-04-09:
> Check the `CALIB` flag while it happens in setup tab.

Screw me. You are right. I see the "CALIBRATING" flag. And if I hold the drone still for a while, the yaw axis works fine after the flag disappears.
[image]

I was aware that the gyro/accelerometer needs to be calibrated first...but I assumed that if roll and pitch were working, that process was complete. Why is the yaw axis the one that does not respond properly during calibration?

closed 2024-05-05 by haslinghuis