betaflight/betaflight-configurator #3891: 2 Profiles for "Power & Battery" Tab

bim4n, 2024-04-11:
### Is your feature request related to a problem? Please describe

When I fly my Long Range copter, the "Power & Battery" Tab is configured for LiPo. But sometimes I use Li-Ion battery and then the config doesn't match for that. A connected buzzer then starts to beep loudly.That would be unnecessary.

### Describe the solution you'd like

For Long Range copter it could be very nice I have **2 Profiles**. 
One Profile for **LiPo**'s and another for **Li-Ion** batteries.

### Describe alternatives you've considered

Manuell config this Tab, but it's not nice ...

chrispijo, 2024-06-05:
I would also like this feature

bim4n, 2024-06-05:
Yes, that would be a very useful Feature !
And I am sure that many pilots would have liked it ...

YarosMallorca, 2024-07-04:
We should look into this, would be nice!

closed 2024-08-12 by github-actions[bot]