betaflight/betaflight-configurator #3893: OSD timer precision mixed up

pfeerick, 2024-04-12:
### Describe the bug

When setting the 'Timer' OSD display element, it appears (unless my understanding of precision is completely muddled up) that the 'Second' and 'Hundredth' precision are swapped?

Additionally, I need to move the timer element between change in precision for it to actually update on the preview or on the VTX feed. 

### To Reproduce

1. Connect to a craft running 4.5.0-RC3, using Configurator 4.5.0-RC3
2. Go to the OSD tab, and with either Timer 1 or Timer 2 display elements, alter the precision, and observe the results. 

### Expected behavior

That Tenths would be Seconds would be `0:00`, tenths would be `0:00.1`, and hundredths would be `0:00.01`.

### Configurator version

10.10.0-RC3 (norevision)

### Flight controller configuration

```txt
# 

# diff

# version
# Betaflight / STM32F411 (S411) 4.5.0 Mar 15 2024 / 10:09:25 (5bf45a2fa) MSP API: 1.46
# config rev: a052dc8

# start the command batch
batch start

board_name CRAZYBEEF4FR
manufacturer_id HAMO

# name: mobula7

# feature
feature -AIRMODE
feature TELEMETRY
feature LED_STRIP
feature OSD

# serial
serial 0 2048 115200 57600 0 115200
serial 1 64 115200 57600 0 115200

# beeper
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# led
led 0 0,0::A:11
led 1 1,0:D:FB:11
led 2 2,0::L:6

# aux
aux 0 0 0 1800 2100 0 0
aux 1 1 1 900 1200 0 0
aux 2 2 1 1300 1700 0 0
aux 3 15 3 1800 2100 0 0
aux 4 28 1 1400 2100 0 0
aux 5 35 2 1800 2100 0 0
aux 6 39 3 1800 2100 0 0

# adjrange
adjrange 0 0 4 1000 2000 12 4 0 0
adjrange 1 0 5 1000 2000 29 5 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 3
vtxtable powervalues 0 1 2
vtxtable powerlabels 25 200 500

# master
set acc_trim_pitch = 2
set acc_calibration = 32,-87,-234,1
set bat_capacity = 250
set pid_process_denom = 2
set osd_warn_bitmask = 264191
set osd_cap_alarm = 200
set osd_tim1 = 1
set osd_tim2 = 2562
set osd_vbat_pos = 2369
set osd_link_quality_pos = 2391
set osd_tim_1_pos = 2423
set osd_tim_2_pos = 2284
set osd_flymode_pos = 2361
set osd_throttle_pos = 2401
set osd_battery_usage_pos = 33002
set osd_flip_arrow_pos = 2339
set osd_stat_bitmask = 132260
set osd_canvas_height = 13
set vtx_band = 4
set vtx_channel = 4
set vtx_power = 2
set vtx_low_power_disarm = ON
set vtx_freq = 5800
set vcd_video_system = NTSC
set craft_name = mobula7
set pilot_name = pfeerick

profile 0

rateprofile 0

# end the command batch
batch end

#
```

haslinghuis, 2024-04-12:
@pfeerick please test the fix.

closed 2024-04-12 by haslinghuis

pfeerick, 2024-04-14:
> @pfeerick please test the fix.

Sorry for the delay... life + shite internet in the way :rofl: ... just finished downloaded latest nightly and yes that fixed it. Simple, but quite annoying :zany_face: 

> just clicking the virtual OSD image will update the element as well, no need to move it.

Oh, thanks for the tip... at least won't have to keep moving stuff back to the right spot :grin:

PR #3894 (2024-04-12): Fix OSD timer
- Fixes #3893