betaflight/betaflight-configurator #3898: VARIO not working anymore on 4.5RC4. It was working fine on RC3

Giammi, 2024-04-18:
### Describe the bug

VARIO not working after upgrading to RC4. Numerical vario in OSD is shown, but stuck to 0.0 m/s. It was working fine on RC3. Barometer is working fine since altitude is given correctly and also in sensor tab altitude sensor is giving consistent results.

### To Reproduce

Flash 4.5RC4 adding VARIO as custom define.

Target Kakute H7

### Expected behavior

Numerical vario to give correct result.

### Configurator version

10.10.0-RC4

### Flight controller configuration

```txt
# 

# diff

# version
# Betaflight / STM32H743 (SH74) 4.5.0 Apr 16 2024 / 08:22:31 (7c8a55258) MSP API: 1.46
# config rev: a826bbb

# start the command batch
batch start

board_name KAKUTEH7
manufacturer_id HBRO

# name: PhoeniX

# resources
resource SERIAL_TX 3 NONE
resource PINIO 1 D08

# feature
feature GPS
feature TELEMETRY
feature OSD
feature ESC_SENSOR

# serial
serial 0 131073 115200 57600 0 115200
serial 1 1 115200 57600 0 115200
serial 3 2 115200 57600 0 115200
serial 5 64 115200 57600 0 115200
serial 6 1024 115200 57600 0 115200

# beeper
beeper -DISARMING
beeper -ARMING_GPS_FIX
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION_FAIL
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# map
map TAER1234

# aux
aux 0 0 1 1750 2100 0 0
aux 1 1 2 1350 1650 0 0
aux 2 46 2 1700 2100 0 0
aux 3 13 2 1350 2100 0 0
aux 4 13 0 1750 2100 0 0
aux 5 13 1 1350 1650 0 0
aux 6 26 1 1350 2100 0 0
aux 7 35 0 1750 2100 0 0
aux 9 40 3 1350 2100 0 0

# rxfail
rxfail 3 s 1450
rxfail 6 s 1500

# master
set gyro_lpf1_static_hz = 0
set gyro_lpf2_static_hz = 300
set dyn_notch_count = 2
set dyn_notch_q = 500
set dyn_notch_min_hz = 60
set gyro_lpf1_dyn_min_hz = 0
set gyro_lpf1_dyn_max_hz = 0
set acc_lpf_hz = 10
set acc_calibration = 3,-118,-207,1
set align_mag = CW90FLIP
set mag_align_pitch = 1800
set mag_align_yaw = 900
set mag_calibration = -100,-186,-49
set rc_smoothing_auto_factor = 35
set rc_smoothing_auto_factor_throttle = 28
set rc_smoothing_setpoint_cutoff = 12
set rc_smoothing_feedforward_cutoff = 12
set rc_smoothing_throttle_cutoff = 20
set fpv_mix_degrees = 25
set blackbox_sample_rate = 1/2
set dshot_idle_value = 800
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_output_reordering = 3,2,1,0,4,5,6,7
set failsafe_delay = 4
set failsafe_procedure = GPS-RESCUE
set align_board_yaw = 180
set vbat_min_cell_voltage = 345
set vbat_warning_cell_voltage = 355
set force_battery_cell_count = 6
set ibata_scale = 147
set yaw_motors_reversed = ON
set small_angle = 180
set mag_declination = 33
set gps_ublox_use_galileo = ON
set gps_set_home_point_once = ON
set gps_rescue_alt_mode = CURRENT_ALT
set gps_rescue_initial_climb = 40
set gps_rescue_ground_speed = 1500
set gps_rescue_max_angle = 55
set gps_rescue_descent_dist = 40
set gps_rescue_throttle_hover = 1350
set gps_rescue_allow_arming_without_fix = ON
set yaw_deadband = 5
set pid_process_denom = 2
set simplified_gyro_filter = OFF
set simplified_gyro_filter_multiplier = 105
set osd_warn_bitmask = 56313
set osd_rssi_dbm_alarm = -115
set osd_cap_alarm = 4000
set osd_alt_alarm = 2000
set osd_ah_max_pit = 50
set osd_ah_max_rol = 90
set osd_tim1 = 1
set osd_vbat_pos = 12335
set osd_rssi_pos = 480
set osd_link_quality_pos = 15554
set osd_link_tx_power_pos = 15522
set osd_rssi_dbm_pos = 3118
set osd_rsnr_pos = 3086
set osd_tim_1_pos = 14509
set osd_tim_2_pos = 12389
set osd_flymode_pos = 14528
set osd_throttle_pos = 3426
set osd_vtx_channel_pos = 373
set osd_crosshairs_pos = 2361
set osd_ah_sbar_pos = 2362
set osd_ah_pos = 2234
set osd_current_pos = 15746
set osd_mah_drawn_pos = 15651
set osd_craft_name_pos = 15363
set osd_gps_speed_pos = 2415
set osd_gps_lon_pos = 2656
set osd_gps_lat_pos = 2624
set osd_gps_sats_pos = 2048
set osd_home_dir_pos = 2330
set osd_home_dist_pos = 2318
set osd_flight_dist_pos = 2254
set osd_compass_bar_pos = 2070
set osd_altitude_pos = 18734
set osd_power_pos = 416
set osd_warnings_pos = 14517
set osd_avg_cell_voltage_pos = 15618
set osd_disarmed_pos = 2391
set osd_nheading_pos = 2062
set osd_up_down_reference_pos = 207
set osd_nvario_pos = 2446
set osd_esc_tmp_pos = 2432
set osd_flip_arrow_pos = 2170
set osd_core_temp_pos = 2400
set osd_stat_bitmask = 42385018
set osd_gps_sats_show_pdop = ON
set osd_camera_frame_width = 30
set osd_camera_frame_height = 16
set debug_mode = GYRO_SCALED
set pinio_box = 40,41,42,43
set rpm_filter_min_hz = 50
set rpm_filter_fade_range_hz = 30
set craft_name = PhoeniX

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 0
set dterm_lpf1_dyn_max_hz = 157
set dterm_lpf1_static_hz = 63
set dterm_lpf2_static_hz = 127
set iterm_relax_cutoff = 5
set pidsum_limit = 1000
set pidsum_limit_yaw = 1000
set yaw_lowpass_hz = 50
set p_pitch = 90
set i_pitch = 64
set d_pitch = 86
set f_pitch = 144
set p_roll = 78
set i_roll = 55
set d_roll = 76
set f_roll = 125
set p_yaw = 78
set i_yaw = 55
set f_yaw = 125
set d_min_roll = 76
set d_min_pitch = 86
set thrust_linear = 10
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 20
set feedforward_jitter_factor = 15
set feedforward_boost = 5
set dyn_idle_min_rpm = 14
set simplified_master_multiplier = 175
set simplified_i_gain = 40
set simplified_d_gain = 145
set simplified_dmax_gain = 0
set simplified_feedforward_gain = 60
set simplified_pitch_pi_gain = 110
set simplified_dterm_filter_multiplier = 85
set tpa_rate = 50
set tpa_breakpoint = 1200

rateprofile 0

# rateprofile 0
set roll_expo = 40
set pitch_expo = 40
set yaw_expo = 40
set roll_srate = 55
set pitch_srate = 55
set yaw_srate = 55

# end the command batch
batch end

#
```

haslinghuis, 2024-04-19:
- Do you have a support ID?
- Try debug `DEBUG_ALTITUDE`

Giammi, 2024-04-19:
@haslinghuis support ID is d9452f24-f4e1-4877-ac3a-cdc7d61da9f6
Attached you can find a log with ALTITUDE logged. It's just a hovering test with some up and downs.
[LOG00012.zip](https://github.com/betaflight/betaflight-configurator/files/15041814/LOG00012.zip)

haslinghuis, 2024-04-19:
Checked on blackbox.betaflight.com but could not get the ALTITUDE debug data.
In the header it says GYRO_SCALED ?

Giammi, 2024-04-19:
@haslinghuis sorry, my bad. I got fooled by the new blackbox interface in the configurator. I selected ALTITUDE in the dropdown list, not in the debug. Now I set it to DEBUG_ALTITUDE. Here you go.
[LOG00013.zip](https://github.com/betaflight/betaflight-configurator/files/15043415/LOG00013.zip)

haslinghuis, 2024-04-19:
Vario works in blackbox. What symbol is shown in OSD?

```
ifdef USE_VARIO
static void osdElementNumericalVario(osdElementParms_t *element)
{
    bool haveBaro = false;
    bool haveGps = false;
#ifdef USE_BARO
    haveBaro = sensors(SENSOR_BARO);
#endif // USE_BARO
#ifdef USE_GPS
    haveGps = sensors(SENSOR_GPS) && STATE(GPS_FIX);
#endif // USE_GPS
    if (haveBaro || haveGps) {
        const float verticalSpeed = osdGetMetersToSelectedUnit(getEstimatedVario()) / 100.0f;
        const char directionSymbol = verticalSpeed < 0 ? SYM_ARROW_SMALL_DOWN : SYM_ARROW_SMALL_UP;
        osdPrintFloat(element->buff, directionSymbol, fabsf(verticalSpeed), "", 1, true, osdGetVarioToSelectedUnitSymbol());
    } else {
        // We use this symbol when we don't have a valid measure
        element->buff[0] = SYM_HYPHEN;
        element->buff[1] = '\0';
    }
}
#endif // USE_VARIO
```

Giammi, 2024-04-19:
@haslinghuis symbol is just the unit of measure m/s with 0.0 before it. I don't see anything wrong with respect to the past. To me looks like it's well represented in OSD, just the value is stuck to 0.0.

haslinghuis, 2024-04-19:
What GPS / MAG unit is used? Try disabling MAG (might have to remove from firmware build for testing)

Giammi, 2024-04-19:
@haslinghuis GPS is a Flywoo Goku V3 M10 with QMC5883L. BTW never mind, I've flashed it again with full chip erase and now it works. So you can close this. Thank you anyway for your time looking into this.

closed 2024-04-19 by haslinghuis