betaflight/betaflight-configurator #3906: 10.10.0-RC4 Missing option to "Permanently enable Expert Mode"

slash128v6, 2024-04-23:
### Describe the bug

In 10.9 there was an option to "Permanently enable Expert Mode" but this appears to be missing in the 10.10 release candidates.

### To Reproduce

Click on options before connecting to a device. No option to "Permanently enable Expert Mode" in the list.

### Expected behavior

In 10.9 in options there was an item to "Permanently enable Expert Mode" in the options list.

### Configurator version

10.10.0-RC4

haslinghuis, 2024-04-23:
Expert mode has been unified and is just a [permanent] switch now

closed 2024-04-23 by haslinghuis

slash128v6, 2024-04-23:
Whoops, I missed that. Sorry for the noise.