betaflight/betaflight-configurator #3907: 10.10.4 Betaflight Configurator MAC-OSX version is Intel only executable

IgnacioHR, 2024-04-23:
### Is your feature request related to a problem? Please describe

Current dmg installers for Apple OSX family is Intel executable but Intel processors are now very old in the Apple world and the new arm64 Mx chips are most common. Also there is a new executable format to contain both versions packaged in the same executable (the Universal executable). 

### Describe the solution you'd like

The request is to produce an Universal version of the dmg installer for OSX

### Describe alternatives you've considered

The alternative is to do nothing. Current Intel version works if you have rosetta installed even if it is not optimal.

IgnacioHR, 2024-04-24:
Hi, sure. I've downloaded 10.10.4-RC4

[image]

And this is a screen capture of the activity monitor
[image]

So, I think, I'm sure about what I'm saying

nerdCopter, 2024-04-24:
Thank you, 
Although highly "beta" at this time, https://app.betaflight.com/ will allow you to configure. it will not allow flashing yet.  Chrome or similar required, will not work in FireFox.  Configurator is moving away from `NW.js` dependent stand-alone app after the full release of 10.10/4.5.

haslinghuis, 2024-05-05:
We are moving away from local installations and NWjs

closed 2024-05-05 by haslinghuis