betaflight/betaflight-configurator #3912: BF Configurator 10.10  :  Some OSD X,Y locations are not saved

wterreb, 2024-04-26:
### Describe the bug

I noticed that some of my OSD items are defaulting to their original position each time I reboot.
More specifically items like GPS Lat, Lng, Craft Name,  etc
Some of the items do seem to get saved correctly.   Maybe its to do with the location of the screen.   For example I noticed it only happens on the items that I try to move to the bottom of the screen.   Items in the top halve of the screen do not seem to have a problem

### To Reproduce

In Betaflight Configurator, go to the OSD tab.   
Choose Video format Auto (I have an analog quad)
Enable GPS Lat and Lng
Change their location to the bottom of the screen
Click save
Reboot and go back to OSD tab.  The locations of these OSD items are now back about two thirds up in the screen.

### Expected behavior

When you move OSD items to the bottom quarter of the screen and click save, then they should be in that same position when you moved them to after a reboot

### Support ID

```txt
e47b6efe-da6f-496d-b44e-10a21a250f46
```

### Flight controller

FLYWOOF745NANO

### Other components

I do not thing it matters as I think the fault is in BF configurator and not the firmware

### How are the different components wired up (including port information)

I do not thing it matters as I think the fault is in BF configurator and not the firmware

### Add any other context about the problem that you think might be relevant here

N/A

haslinghuis, 2024-04-26:
Support ID is not working

[e47b6efe-da6f-496d-b44e-10a21a250f46](https://build.betaflight.com/api/support/e47b6efe-da6f-496d-b44e-10a21a250f46)

haslinghuis, 2024-04-26:
PAL or NTSC ?

closed 2024-05-05 by haslinghuis