betaflight/betaflight-configurator #3918: Angle mode not going yellow but horizon is?

Afronutella, 2024-04-29:
### Describe the bug

i just upgraded my build to 4.5 from 4.1, i did a clean build but the angle mode will not go yellow for some reason. so i tried rolling back the release to 4.4.3 but that still did nothing. am i missing something? (acc is on and calibrated, everything eles is on default). 

[image]
[image]
[image]
[image]

### To Reproduce

just flash and go to modes 

### Expected behavior

for angle to go yellow when pressed

### Configurator version

4.5 and 4.4.3

### Flight controller configuration

```txt
# version
# Betaflight / STM32F405 (S405) 4.5.0 Apr 28 2024 / 02:32:10 (c155f5830) MSP API: 1.46
# config rev: 3068e6e

# start the command batch
batch start

board_name OMNIBUSF4SD
manufacturer_id AIRB

# feature
feature INFLIGHT_ACC_CAL
feature SOFTSERIAL
feature TELEMETRY
feature LED_STRIP
feature OSD

# serial
serial 5 64 115200 57600 0 115200

# aux
aux 0 0 1 1825 2100 0 0
aux 1 1 2 1800 2100 0 0
aux 2 2 2 1400 1650 0 0
aux 3 13 4 1700 2100 0 0
aux 4 35 3 1800 2100 0 0

# master
set acc_calibration = 13,-23,55,1

profile 0

rateprofile 0

# end the command batch
batch end
```

nerdCopter, 2024-04-29:
mine works when i click `save`.
[image]
[image]

i would have asked if accel is enabled, but i see it is.  is it calibrated as well?  
doubtful, but maybe is your arming angle not sufficient?
[image]

Afronutella, 2024-04-29:
Its enabled and calibrated. Not sure why its not working tho. Arming angle is 25 and ive tried flipping the drone while the button is on. But if that was the problem then horizon should also not activate. 

Think next im gonna roll it back at 4.2/4.3 and see if its the FC thats the issue. 

Also i checked the dump and its registering in AUX. So im stumped ngl.

Afronutella, 2024-04-29:
Yea i cant figure out why this is happening. If anyone has any ideas or thoughts it would be greatly appreciated. 

Also on the modes section angle is a lighter shade of grey for some reason as if its not been used. But it obviously has

haslinghuis, 2024-04-29:
:confused: Works here as expected

Afronutella, 2024-04-29:
Damn ill just have to use horizon as angle if i ever need and just change the strength and trans. Ty tho

NaorFPV, 2024-05-06:
i have the same issue, airmode also will not light yellow.. with BF4.5 release

SteveCEvans, 2024-05-14:
Using the MSP radio emulator so you can see the AUX 1 position on screen, see below. It's working as it should.

[image]

[image]

[image]

Am I missing something?

NaorFPV, 2024-05-14:
[image]
[image]

haslinghuis, 2024-05-14:
@NaorFPV try light mode and it works.

Found the issue started in be0103f

@McGiverGim please have a look at it.

Afronutella, 2024-05-14:
yup perfect it works on light mode only. ty guys 
[image]

Afronutella, 2024-05-14:
and works with the fix on dark 
[image]

closed 2024-05-14 by Afronutella